analogdevicesinc/libiio #331: No Pluto found with libiio on Ubunutu 18.04

patel999jay, 2020-01-04:
No pluto/context has been found on Ubunutu 18.04. it found pluto on usb. I built libiio from source but still same problem.

 `jay@jay-MS-7885:~$ iio_info -s
Library version: 0.18 (git tag: v0.18)
Compiled with backends: local xml ip
No contexts found.`

pcercuei, 2020-01-04:
It says ```compiled with backends: local xml ip```, so it was not compiled with the USB backend.

patel999jay, 2020-01-04:
I already had a look on the source code and CMakeLists.txt, but didn't find a solution. Is the 'WITH_IIOD_USBD' option is only available under certain conditions ?

pcercuei, 2020-01-04:
You need the ```libusb-1.0.0-dev``` package installed, then the option ```WITH_USB_BACKEND``` should be available.

patel999jay, 2020-01-05:
Thanks. I had already installed `libusb-dev` And build this from source but not sure why it didn’t build with usb backend support.

pcercuei, 2020-01-05:
Libiio uses libusb-1.0, the ```libusb-dev``` is for libusb-0.x, you are not the first one to fall for this trap, believe me ;)

closed 2020-01-05 by patel999jay