analogdevicesinc/libiio #335: Synchronizing multiple buffered devices

fpagliughi, 2020-01-10:
I have an application that will have two or more relatively low-speed ADC chips (~50kHz), with the plan to drive them both from the same trigger using buffered input. My hope is to try to match the buffers so that the first sample in each buffer corresponds to ones obtained from the same trigger for each device. Or as close as possible.

My initial test shows that the second buffer is starting dozens of samples after the first.

So a few questions:

1. It seems that buffers are automatically enabled on creation. Is this true? If so, then it's likely that the lag is due to the time to create the second buffer?

2. Is there a way to enable and disable buffers outside creation and destruction? It seems that there is a hidden device attribute, "buffer/enable" that will do this. If so, why is this hidden, and not part of the public API, like `int  iio_buffer_enable(struct iio_buffer *buf, bool enable);` ?

3. Is there any other "trick" to help with this?

I'm testing with a network context, but it will be a local context in the final application.
Thanks!

pcercuei, 2020-01-10:
Yes, buffers are enabled on creation, and that's most likely the cause of your lag.

The buffer/enable attribute is written when the buffer object is created. There is no way to manually enable or disable a buffer, and even if there was, it wouldn't fix your issue, since there would still be a short time between two writes to enable each ADC.

If you want to synchronize the captures, make sure that your trigger gets active only after the two buffers have been created.

dsp-payloads, 2020-01-10:
Could the moderator please help me unsubscribe as still getting emails from
this thread after multiple unsubscription requests. Appologies for the
global notice. Thanks.

pcercuei, 2020-01-10:
@dsp-payloads in the thread (which is https://github.com/analogdevicesinc/libiio/issues/335) there is a subscribe/unsubscribe button for the notifications, maybe you hit it by mistake?

fpagliughi, 2020-01-23:
That worked. Thanks @pcercuei.  I actually don't have access to the enable bit of the timer, but I can change the sampling frequency. So I set it _really_ slow, create the buffer, and then set it to the actual frequency. I probably need to throw away the 1st buffer, but that's no problem.

Just out of curiosity, though, why is the buffer enable/disable hidden? Is there a reason not to use it?

pcercuei, 2020-01-23:
Whether the interface is low-speed (read/write on the IIO dev node) or high-speed (ioctls to dequeue buffers), the buffer/enable attribute has to be enabled at different moments, so the best way to have one API that works on both modes was to have the parameter managed internally.

rgetz, 2020-03-31:
Closing, since I think this has been answered.

closed 2020-03-31 by rgetz