analogdevicesinc/libiio #336: warnings on some arch

rgetz, 2020-01-11:
based on debian testing, on (kfreebsd-amd64) 
https://buildd.debian.org/status/fetch.php?pkg=libiio&arch=kfreebsd-amd64&ver=0.18-1&stamp=1565884272&raw=0

```
/usr/bin/cc -DLIBIIO_EXPORTS=1 -DMATLAB_BINDINGS_API=1 -DNETWORK_BACKEND=1 -DSERIAL_BACKEND=1 -DXML_BACKEND=1 -D_POSIX_C_SOURCE=200809L -D__XSI_VISIBLE=500 -Diio_EXPORTS -I/<<PKGBUILDDIR>> -I/<<PKGBUILDDIR>>/build -I/usr/include/libxml2  -g -O2 -fdebug-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fvisibility=hidden -Wall -Wextra -Wno-unused-parameter -Wno-sign-compare -Wpedantic -fPIC   -std=c99 -o CMakeFiles/iio.dir/network.c.o   -c /<<PKGBUILDDIR>>/network.c
/<<PKGBUILDDIR>>/network.c: In function ‘create_cancel_fd’:
/<<PKGBUILDDIR>>/network.c:240:8: warning: implicit declaration of function ‘pipe2’; did you mean ‘pipe’? [-Wimplicit-function-declaration]
  ret = pipe2(io_ctx->cancel_fd, O_CLOEXEC | O_NONBLOCK);
        ^~~~~
        pipe
```

based on the freebsd pipe2 man page,https://www.freebsd.org/cgi/man.cgi?pipe(2) 
```
       #include <unistd.h>
       int pipe2(int pipefd[2], int flags);
```

which we are - but we are still getting the warning...

For Linux - pipe2 only shows up with _GNU_SOURCE, which we only do on Linux, not on freebsd (so I'm not familiar enough with freebsd to know if that is the problem or not).

-Robin

rgetz, 2020-01-16:
OK - this specific bug report is on Debian GNU/kFreeBSD, which is a port that consists of GNU userland using the GNU C library on top of FreeBSD's kernel, coupled with the regular Debian package set; and is no longer an officially supported Debian architecture.
https://www.debian.org/ports/kfreebsd-gnu/

FreeBSD works fine - and builds without changes (except for not including the bindings?)
https://www.freebsd.org/cgi/ports.cgi?query=libiio&stype=all

I will install FreeBSD, and try to see when they don't build the bindings, but other than that - I think this is non a bug.

rgetz, 2020-01-16:
Looks like FreeBSD 12 puts libusb in a different directory, I needed to do:

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 552dcf7..0f4a218 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -163,8 +163,13 @@ if(WITH_LOCAL_BACKEND)
        endif()
 endif()
 
-find_library(LIBUSB_LIBRARIES usb-1.0)
-find_path(LIBUSB_INCLUDE_DIR libusb-1.0/libusb.h)
+if (CMAKE_SYSTEM_NAME STREQUAL "FreeBSD")
+       find_library(LIBUSB_LIBRARIES usb)
+       find_path(LIBUSB_INCLUDE_DIR libusb.h)
+else()
+       find_library(LIBUSB_LIBRARIES usb-1.0)
+       find_path(LIBUSB_INCLUDE_DIR libusb-1.0/libusb.h)
+endif()
 if (LIBUSB_LIBRARIES AND LIBUSB_INCLUDE_DIR)
        message(STATUS "Looking for libusb-1.0 : Found")
        option(WITH_USB_BACKEND "Enable the libusb backend" ON)
diff --git a/usb.c b/usb.c
index 141bd41..c8a3b81 100644
--- a/usb.c
+++ b/usb.c
@@ -22,7 +22,11 @@
 
 #include <ctype.h>
 #include <errno.h>
+#ifdef __FreeBSD__
+#include <libusb.h>
+#else
 #include <libusb-1.0/libusb.h>
+#endif
 #include <stdbool.h>
 #include <string.h>
```

to get it to build with USB...

rgetz, 2020-01-16:
I still get this warning on FreeBSD

```
[ 37%] Building C object CMakeFiles/iio.dir/network.c.o
/root/github/libiio/network.c:240:8: warning: implicit declaration of function 'pipe2' is invalid in C99 [-Wimplicit-function-declaration]
        ret = pipe2(io_ctx->cancel_fd, O_CLOEXEC | O_NONBLOCK);
              ^
```

to resolve that, I need to do:

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 552dcf7..de64eeb 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -97,6 +97,10 @@ IF(${CMAKE_SYSTEM_NAME} MATCHES "Linux")
        set(CMAKE_REQUIRED_DEFINITIONS "-D_GNU_SOURCE=1")
        add_definitions(-D_GNU_SOURCE=1)
 endif()
+if(${CMAKE_SYSTEM_NAME} MATCHES "FreeBSD")
+       set(CMAKE_REQUIRED_DEFINITIONS "-D__BSD_VISIBLE")
+       add_definitions(-D__BSD_VISIBLE=1)
+endif()
```

rgetz, 2020-01-16:
I confirmed with Yuri (maintainer for libiio in FreeBSD https://www.freshports.org/misc/libiio ), that this was the correct approach for the pipe2 warning, so made a pull request for that.

The libusb might be a little more complex - so I that will be a separate pull request.

pcercuei, 2020-01-16:
Looks good to me.

rgetz, 2020-01-17:
Ok - these are in process in the different pull requests, so close this.

mhennerich, 2020-01-28:
ok - the fix in 645391bd9c0ac3adfef377e3ce8c7231ac90b235 causes a failure on my Ubuntu 16.04.6 LTS. 

michael@mhenneri-D06:~/devel/git/libiio$ cmake .
-- Looking for libusb-1.0 : Found
CMake Warning at CMakeLists.txt:213 (message):
  The installed version of libserialport is too old.  The minimum version
  supported is 0.1.1.  Disabling Serial support.

-- Building with Network back end support
-- Building with Avahi, a zero-configuration networking (zeroconf) implementation
-- Found Python: Building bindings
-- Check for case-sensitive file systems
-- File system is case-sensitive
-- Configuring done
-- Generating done
-- Build files have been written to: /home/michael/devel/git/libiio
michael@mhenneri-D06:~/devel/git/libiio$ make
Scanning dependencies of target iio
[  2%] Building C object CMakeFiles/iio.dir/backend.c.o
[  5%] Building C object CMakeFiles/iio.dir/channel.c.o
[  7%] Building C object CMakeFiles/iio.dir/device.c.o
[ 10%] Building C object CMakeFiles/iio.dir/context.c.o
[ 13%] Building C object CMakeFiles/iio.dir/buffer.c.o
[ 15%] Building C object CMakeFiles/iio.dir/utilities.c.o
[ 18%] Building C object CMakeFiles/iio.dir/scan.c.o
[ 21%] Building C object CMakeFiles/iio.dir/sort.c.o
[ 23%] Building C object CMakeFiles/iio.dir/local.c.o
[ 26%] Building C object CMakeFiles/iio.dir/usb.c.o
/home/michael/devel/git/libiio/usb.c:25:20: fatal error: libusb.h: No such file or directory
compilation terminated.
CMakeFiles/iio.dir/build.make:278: recipe for target 'CMakeFiles/iio.dir/usb.c.o' failed
make[2]: *** [CMakeFiles/iio.dir/usb.c.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/iio.dir/all' failed
make[1]: *** [CMakeFiles/iio.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
michael@mhenneri-D06:~/devel/git/libiio$

rgetz, 2020-01-28:
Ok - will spin up a vm, and try it out....

mhennerich, 2020-01-28:
I have both versions of libusb installed on my system:
libusb-dev and libusb-1.0-0-dev

```
michael@mhenneri-D06:~/devel/git/libiio$ dpkg-query -l libusb-1.0-0-dev
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                   Version          Architecture     Description
+++-======================-================-================-==================================================
ii  libusb-1.0-0-dev:amd64 2:1.0.20-1       amd64            userspace USB programming library development file
```
```
michael@mhenneri-D06:~/devel/git/libiio$ dpkg-query -l libusb-dev
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                   Version          Architecture     Description
+++-======================-================-================-==================================================
ii  libusb-dev             2:0.1.12-28      amd64            userspace USB programming library development file
michael@mhenneri-D06:~/devel/git/libiio$ 
```
Restoring the previous include path fixes it

```
michael@mhenneri-D06:~/devel/git/libiio$ git diff
diff --git a/usb.c b/usb.c
index 49af0cc..141bd41 100644
--- a/usb.c
+++ b/usb.c
@@ -22,7 +22,7 @@
 
 #include <ctype.h>
 #include <errno.h>
-#include <libusb.h>
+#include <libusb-1.0/libusb.h>
 #include <stdbool.h>
 #include <string.h>
```

pcercuei, 2020-01-28:
I do have both installed too, and it builds fine here.
Check in the CMake GUI (or the CMakeCache.txt) what's the value of LIBUSB_INCLUDE_DIR, it should be something like /usr/include/libusb-1.0.

mhennerich, 2020-01-28:
michael@mhenneri-D06:~/devel/git/libiio$ grep -R LIBUSB_INCLUDE_DIR
CMakeCache.txt:LIBUSB_INCLUDE_DIR:PATH=/usr/include

After: 
michael@mhenneri-D06:~/devel/git/libiio$ git clean -f -d -x

michael@mhenneri-D06:~/devel/git/libiio$ grep -R LIBUSB_INCLUDE_DIR
CMakeCache.txt:LIBUSB_INCLUDE_DIR:PATH=/usr/include/libusb-1.0

my fault - sorry for the noise - closed

closed 2020-01-28 by mhennerich

rgetz, 2020-01-29:
OK  - thanks for double checking, and letting us know.. (I was puzzled, since we build on Ubunut 16 on Travis-CI still).

BTW - Ubuntu 16.04 LTS "End of Standard Support" is April 2021, time to upgrade...

commit 100f9101 (2020-01-17): fix #336, remove warnings on FreeBSD

Previously, building libiio on FreeBSD would warn with:
network.c:240:8: warning: implicit declaration of function 'pipe2' is invalid in C99

This fixes that for all BSDs, since pipe2 (inside <unistd.h>) is protected by __BSD_VISIBLE.
On Linux, it's protected by _GNU_SOURCE (which is already defined in the cmake).
commit 645391bd (2020-01-17): Fix #336, make sure libusb can be found on FreeBSD variants

FreeBSD doesn't call it libusb-1,0 like linux does - it calls it libusb
and just installs it into /usr/include/*, not /usr/include/libusb-1.0/*
like Linux, Apple and Windows does... so handle these differences.

Tested on FreeBSD 12.1 (Release) & Debian 10 (Buster)
commit a92ecf32 (2020-01-20): Merge pull request #343 from analogdevicesinc/rgetz-patch-2

fix #336, remove warnings on FreeBSD
commit ad53b3f5 (2020-01-20): Merge pull request #346 from analogdevicesinc/rgetz-fix-freebsd-usb

Fix #336, make sure libusb can be found on FreeBSD variants