analogdevicesinc/libiio #344: drop support for python 2?

rgetz, 2020-01-16:
I was looking in upstream debian, they apply this patch:

```
--- a/bindings/python/CMakeLists.txt
+++ b/bindings/python/CMakeLists.txt
@@ -1,9 +1,10 @@
-cmake_minimum_required(VERSION 2.8.7)
+cmake_minimum_required(VERSION 3.12)
 project(libiio-py NONE)
 
-include(FindPythonInterp)
+find_package (Python2 COMPONENTS Interpreter Development)
+find_package (Python3 COMPONENTS Interpreter Development)
 
-if (PYTHONINTERP_FOUND)
+if (Python2_Interpreter_FOUND)
 	option(PYTHON_BINDINGS "Install Python bindings" ON)
 
 	if (PYTHON_BINDINGS)
@@ -15,10 +16,30 @@
 
 		configure_file(${CMAKE_CURRENT_SOURCE_DIR}/iio.py  ${CMAKE_CURRENT_BINARY_DIR}/iio.py COPYONLY)
 
-		add_custom_target(libiio-py ALL DEPENDS ${SETUP_PY} COMMAND ${PYTHON_EXECUTABLE} ${SETUP_PY} --quiet build)
+		add_custom_target(libiio-py ALL DEPENDS ${SETUP_PY} COMMAND ${Python2_EXECUTABLE} ${SETUP_PY} --quiet build)
 
 		if(NOT SKIP_INSTALL_ALL)
-			install(CODE "execute_process(WORKING_DIRECTORY ${CMAKE_CURRENT_BINARY_DIR} COMMAND ${PYTHON_EXECUTABLE} ${SETUP_PY} install --root=\$ENV{DESTDIR}/ --prefix=${CMAKE_INSTALL_PREFIX})")
+			install(CODE "execute_process(WORKING_DIRECTORY ${CMAKE_CURRENT_BINARY_DIR} COMMAND ${Python2_EXECUTABLE} ${SETUP_PY} install --root=\$ENV{DESTDIR}/ --prefix=${CMAKE_INSTALL_PREFIX})")
+		endif()
+	endif()
+endif()
+
+if (Python3_Interpreter_FOUND)
+	option(PYTHON3_BINDINGS "Install Python3 bindings" ON)
+
+	if (PYTHON3_BINDINGS)
+
+		set(SETUP_PY_IN ${CMAKE_CURRENT_SOURCE_DIR}/setup.py.cmakein)
+		set(SETUP_PY ${CMAKE_CURRENT_BINARY_DIR}/setup.py)
+
+		configure_file(${SETUP_PY_IN} ${SETUP_PY})
+
+		configure_file(${CMAKE_CURRENT_SOURCE_DIR}/iio.py  ${CMAKE_CURRENT_BINARY_DIR}/iio.py COPYONLY)
+
+		add_custom_target(libiio-py3 ALL DEPENDS ${SETUP_PY} COMMAND ${Python3_EXECUTABLE} ${SETUP_PY} --quiet build)
+
+		if(NOT SKIP_INSTALL_ALL)
+			install(CODE "execute_process(WORKING_DIRECTORY ${CMAKE_CURRENT_BINARY_DIR} COMMAND ${Python3_EXECUTABLE} ${SETUP_PY} install --root=\$ENV{DESTDIR}/ --prefix=${CMAKE_INSTALL_PREFIX})")
 		endif()
 	endif()
 endif()
```

I know python 2 is disappearing from many distributions, or they are including python2 and python3, (I think libiio finds python 2 by default), do we need to do anything here?

Thanks

matejk, 2020-01-17:
Module FindPython (cmake >= 3.12) prefers Python 3.

https://cmake.org/cmake/help/latest/module/FindPython.html

Recent changes of cmake python binding build files use that module if recent version of cmake is installed so it shall properly use Python 3 by default on recent distributions.

It is possible to override the default like this:

```
cmake -DPYTHON_EXECUTABLE:FILEPATH=/usr/bin/python ../libiio/
```

rgetz, 2020-01-17:
That is what I wasn't sure - thanks for feedback / confirmation.

closed 2020-01-17 by rgetz