analogdevicesinc/libiio #350: License issue

rgetz, 2020-01-21:
This was pointed out in : 
https://bugzilla.redhat.com/show_bug.cgi?id=1426465

In all of the libiio *.c files (backend.c  buffer.c  channel.c  context.c  device.c  iiod-client.c  local.c  lock.c  network.c  scan.c  serial.c  sort.c  usb.c  utilities.c  xml.c), it's indicated:
```
 * This library is free software; you can redistribute it and/or
 * modify it under the terms of the GNU Lesser General Public
 * License as published by the Free Software Foundation; either
 * version 2.1 of the License, or (at your option) any later version.
```
In mainpage.dox it indicates:
```
Libiio has been developed and is released under the terms of the GNU Lesser General Public
License, version 2. 
```
which is missing the "any later version". Same for the README.md file.

Also, the utilities, which are in the `./test` directory, aren't a library, and LGPL doesn't really make sense for those applications. Same for the `./examples` directory. Section 2a of the LGPL states:
https://www.gnu.org/licenses/old-licenses/lgpl-2.1.en.html
```
a) The modified work must itself be a software library.
```

I would propose that:
-  I fix the README and Dox to match the license in the code (which I will do).
- the files in the ./test directory be updated to "GPLv2 or any later version"
- the files in the ./examples directory be updated to "GPLv2 or any later version"

I just need everyone's OK for 2 & 3 (if anyone doesn't answer, I won't do it).

Authors of c files in the ./test directory: @johnnykv @julienmalik @larsclausen @lucasrangit @mtitinger @mfornero @mhei @mhennerich @pcercuei @rgetz : I need your permission to do this.

Authors of c files in the ./examples directory @dNechita @dimasad @geertu @larsclausen @lucasrangit @mhennerich @pcercuei @SrikanthPagadarai @tfcollins : I need your permission to do this. 

Thanks

johnnykv, 2020-01-21:
Permission hereby given.

mhennerich, 2020-01-21:
Hereby approved

geertu, 2020-01-21:
Approved.

dNechita, 2020-01-21:
I'm OK with it

pcercuei, 2020-01-21:
Sure, approved.

mtitinger, 2020-01-21:
Ok with me,
cheers!
M.

larsclausen, 2020-01-21:
ok

dimasad, 2020-01-21:
I agree.

lucasrangit, 2020-01-21:
OK

mfornero, 2020-01-22:
Approved

tfcollins, 2020-01-22:
Approve

SrikanthPagadarai, 2020-01-22:
I'm ok with it.

mhei, 2020-01-22:
Sure I agree, thanks for digging into this!

rgetz, 2020-01-22:
So that just leaves @julienmalik - I will reach out via email and get permission for this patch:

https://github.com/analogdevicesinc/libiio/commit/ac8debcedda20a7713bfa13411d7cd916dc5295a#diff-819036057d9192362290c6c2235e15ab

julienmalik, 2020-01-23:
I hereby agree on the license change to "GPLv2 or any later version".

rgetz, 2020-01-23:
Thanks everyone - will work on this later today.

rgetz, 2020-01-28:
closing this since there is a pull request in progress.

closed 2020-01-28 by rgetz

rgetz, 2020-02-03:
@dNechita pointed out the binding examples needed the same love.

libiio/bindings/python/examples were all done by @pcercuei
libiio/bindings/csharp/examples were all done by @pcercuei 
libiio/bindings/matlab was done by @tfcollins and @radhermit (but are weird since they are licensed under the ADI-BSD license...)

I know that @pcercuei and @tfcollins said it was OK above, that leaves Tim...

Thanks
-Robin

radhermit, 2020-02-03:
> I know that @pcercuei and @tfcollins said it was OK above, that leaves Tim...

It's fine by me.

rgetz, 2020-02-04:
Thanks - will update.

commit f8cd903f (2020-01-29): Licences: fix #350, and clean up license.

This ensures that applications are released under a license appropriate
for applications (GPL).
And doc (readme & Dox) ensures that everything is clear.
commit 43c1df08 (2020-02-02): Licences: fix #350, and clean up license.

This ensures that applications are released under a license appropriate
for applications (GPL).
And doc (readme & Dox) ensures that everything is clear.
commit 30c4728f (2020-02-02): Licences: fix #350, and clean up license.

This ensures that applications are released under a license appropriate
for applications (GPL).
And doc (readme & Dox) ensures that everything is clear.
commit 6a777b70 (2020-02-04): Licences: fix #350, and clean up license.

This ensures that applications are released under a license appropriate
for applications (GPL).
And doc (readme & Dox) ensures that everything is clear.