analogdevicesinc/libiio #364: review CSharp bindings

rgetz, 2020-02-05:
This is a list of functions listed as __api in iio.h, and the number of times they show up in the C# bindings.

```
0 iio_buffer_cancel
0 iio_buffer_first
0 iio_buffer_foreach_sample
0 iio_buffer_get_data
0 iio_buffer_get_device
0 iio_buffer_get_poll_fd
0 iio_buffer_set_blocking_mode
0 iio_buffer_set_data
0 iio_buffer_step
0 iio_channel_attr_read_all
0 iio_channel_attr_read_bool
0 iio_channel_attr_read_double
0 iio_channel_attr_read_longlong
0 iio_channel_attr_write_all
0 iio_channel_attr_write_bool
0 iio_channel_attr_write_double
0 iio_channel_attr_write_longlong
0 iio_channel_attr_write_raw
0 iio_channel_convert
0 iio_channel_convert_inverse
0 iio_channel_find_attr
0 iio_channel_get_device
0 iio_channel_get_index
0 iio_channel_get_modifier
0 iio_channel_get_type
0 iio_channel_set_data
0 iio_context_find_device
0 iio_context_get_attr
0 iio_context_get_attrs_count
0 iio_context_get_attr_value
0 iio_context_info_get_description
0 iio_context_info_get_uri
0 iio_context_info_list_free
0 iio_create_local_context
0 iio_create_scan_context
0 iio_create_xml_context
0 iio_create_xml_context_mem
0 iio_device_attr_read_all
0 iio_device_attr_read_bool
0 iio_device_attr_read_double
0 iio_device_attr_read_longlong
0 iio_device_attr_write_all
0 iio_device_attr_write_bool
0 iio_device_attr_write_double
0 iio_device_attr_write_longlong
0 iio_device_attr_write_raw
0 iio_device_buffer_attr_read
0 iio_device_buffer_attr_read_all
0 iio_device_buffer_attr_read_bool
0 iio_device_buffer_attr_read_double
0 iio_device_buffer_attr_read_longlong
0 iio_device_buffer_attr_write
0 iio_device_buffer_attr_write_all
0 iio_device_buffer_attr_write_bool
0 iio_device_buffer_attr_write_double
0 iio_device_buffer_attr_write_longlong
0 iio_device_buffer_attr_write_raw
0 iio_device_debug_attr_read_all
0 iio_device_debug_attr_read_bool
0 iio_device_debug_attr_read_double
0 iio_device_debug_attr_read_longlong
0 iio_device_debug_attr_write_all
0 iio_device_debug_attr_write_bool
0 iio_device_debug_attr_write_double
0 iio_device_debug_attr_write_longlong
0 iio_device_debug_attr_write_raw
0 iio_device_find_attr
0 iio_device_find_buffer_attr
0 iio_device_find_channel
0 iio_device_find_debug_attr
0 iio_device_get_buffer_attr
0 iio_device_get_buffer_attrs_count
0 iio_device_get_context
0 iio_device_get_data
0 iio_device_identify_filename
0 iio_device_set_data
0 iio_device_set_kernel_buffers_count
0 iio_get_backend
0 iio_get_backends_count
0 iio_has_backend
0 iio_scan_context_destroy
0 iio_scan_context_get_info_list
0 iio_strerror
2 iio_buffer_destroy
2 iio_buffer_push
2 iio_buffer_push_partial
2 iio_buffer_refill
2 iio_channel_attr_get_filename
2 iio_channel_attr_read
2 iio_channel_attr_write
2 iio_channel_disable
2 iio_channel_enable
2 iio_channel_get_attrs_count
2 iio_channel_get_data
2 iio_channel_get_data_format
2 iio_channel_get_id
2 iio_channel_get_name
2 iio_channel_is_enabled
2 iio_channel_is_output
2 iio_channel_is_scan_element
2 iio_channel_read_raw
2 iio_channel_write_raw
2 iio_context_clone
2 iio_context_destroy
2 iio_context_get_description
2 iio_context_get_devices_count
2 iio_context_get_name
2 iio_context_get_version
2 iio_context_get_xml
2 iio_context_set_timeout
2 iio_create_context_from_uri
2 iio_create_default_context
2 iio_create_network_context
2 iio_device_attr_read
2 iio_device_attr_write
2 iio_device_create_buffer
2 iio_device_debug_attr_read
2 iio_device_debug_attr_write
2 iio_device_get_attrs_count
2 iio_device_get_channels_count
2 iio_device_get_debug_attrs_count
2 iio_device_get_id
2 iio_device_get_name
2 iio_device_get_sample_size
2 iio_device_get_trigger
2 iio_device_is_trigger
2 iio_device_reg_read
2 iio_device_reg_write
2 iio_device_set_trigger
2 iio_library_get_version
3 iio_buffer_end
4 iio_channel_get_attr
4 iio_channel_read
4 iio_channel_write
4 iio_context_get_device
4 iio_device_get_attr
4 iio_device_get_channel
4 iio_device_get_debug_attr
5 iio_buffer_start
```

We should have a quick look, and make sure all the things that should be exposed are. (For example, the python bindings also expose more functions.

Ben3eeE, 2020-02-20:
This seems like a great project for a dotnet core application I am working on, thank you for making it available 🙇 The current bindings seems like they are made for dotnet framework and use the dll file. Have you considered to make it available for dotnet core easily?

dNechita, 2020-02-20:
We will look into that. Thank you for suggesting it!

rgetz, 2020-03-30:
In doing some of the reformatting as part of #413 , [Codacy](https://app.codacy.com/manual/commodo/libiio/pullRequest?prid=5265101) had alot of hints to make this "better"?

I don't know if we want to look at these and resolve them all? @dNechita ? @adisuciu ? I don't know how many of the suggestions would effect the API.

dNechita, 2020-03-31:
I will take into account the things that Codacy reports when I will review these bindings.

rgetz, 2020-04-16:
Applying same logic as the python bindings (which makes sense). and skipping all the `_all`, `_bool` `_longlong`, etc (since it's easy to cast those).

```
for funct in $(grep ^__api iio.h | tr " " "\n" | grep "(" | sed 's/(.*$//' | grep -v -e _bool$ -e _double$ -e _longlong$ -e _all$ -e _raw$);do if [ $(grep -c $funct ./bindings/csharp/*.cs -l | wc -l) -eq "0" ] ; then echo $funct missing ; fi ; done
```

gives us:
```
iio_create_scan_context missing
iio_scan_context_destroy missing
iio_scan_context_get_info_list missing
iio_context_info_list_free missing
iio_context_info_get_description missing
iio_context_info_get_uri missing
iio_strerror missing
iio_has_backend missing
iio_get_backends_count missing
iio_get_backend missing
iio_create_local_context missing
iio_create_xml_context missing
iio_create_xml_context_mem missing
iio_context_get_attrs_count missing
iio_context_get_attr missing
iio_context_get_attr_value missing
iio_context_find_device missing
iio_device_get_context missing
iio_device_get_buffer_attrs_count missing
iio_device_get_buffer_attr missing
iio_device_find_channel missing
iio_device_find_attr missing
iio_device_find_buffer_attr missing
iio_device_buffer_attr_read missing
iio_device_buffer_attr_write missing
iio_device_set_data missing
iio_device_get_data missing
iio_device_set_kernel_buffers_count missing
iio_channel_get_device missing
iio_channel_find_attr missing
iio_channel_set_data missing
iio_channel_get_type missing
iio_channel_get_modifier missing
iio_buffer_get_device missing
iio_buffer_get_poll_fd missing
iio_buffer_set_blocking_mode missing
iio_buffer_cancel missing
iio_buffer_first missing
iio_buffer_step missing
iio_buffer_foreach_sample missing
iio_buffer_set_data missing
iio_buffer_get_data missing
iio_channel_get_index missing
iio_channel_convert missing
iio_channel_convert_inverse missing
iio_device_find_debug_attr missing
iio_device_identify_filename missing
```

which is a more manageable list...

cristi-iacob, 2020-04-27:
I will take a look on this.

rgetz, 2020-06-04:
This was merged, so closing.

closed 2020-06-04 by rgetz