analogdevicesinc/libiio #365: review python bindings

rgetz, 2020-02-05:
Just need to look at this with a more critical I than mine. These are the times that the functions defined as __api in iio.h show up in the python bindings. 0 means it is missing.

```
0 iio_buffer_cancel
0 iio_buffer_first
0 iio_buffer_foreach_sample
0 iio_buffer_get_data
0 iio_buffer_get_device
0 iio_buffer_get_poll_fd
0 iio_buffer_set_blocking_mode
0 iio_buffer_set_data
0 iio_buffer_step
0 iio_channel_attr_read_all
0 iio_channel_attr_read_bool
0 iio_channel_attr_read_double
0 iio_channel_attr_read_longlong
0 iio_channel_attr_write_all
0 iio_channel_attr_write_bool
0 iio_channel_attr_write_double
0 iio_channel_attr_write_longlong
0 iio_channel_attr_write_raw
0 iio_channel_convert
0 iio_channel_convert_inverse
0 iio_channel_find_attr
0 iio_channel_get_data
0 iio_channel_get_data_format
0 iio_channel_get_device
0 iio_channel_get_index
0 iio_channel_get_modifier
0 iio_channel_get_type
0 iio_channel_set_data
0 iio_context_find_device
0 iio_context_get_attr_value
0 iio_create_xml_context_mem
0 iio_device_attr_read_all
0 iio_device_attr_read_bool
0 iio_device_attr_read_double
0 iio_device_attr_read_longlong
0 iio_device_attr_write_all
0 iio_device_attr_write_bool
0 iio_device_attr_write_double
0 iio_device_attr_write_longlong
0 iio_device_attr_write_raw
0 iio_device_buffer_attr_read
0 iio_device_buffer_attr_read_all
0 iio_device_buffer_attr_read_bool
0 iio_device_buffer_attr_read_double
0 iio_device_buffer_attr_read_longlong
0 iio_device_buffer_attr_write
0 iio_device_buffer_attr_write_all
0 iio_device_buffer_attr_write_bool
0 iio_device_buffer_attr_write_double
0 iio_device_buffer_attr_write_longlong
0 iio_device_buffer_attr_write_raw
0 iio_device_debug_attr_read_all
0 iio_device_debug_attr_read_bool
0 iio_device_debug_attr_read_double
0 iio_device_debug_attr_read_longlong
0 iio_device_debug_attr_write_all
0 iio_device_debug_attr_write_bool
0 iio_device_debug_attr_write_double
0 iio_device_debug_attr_write_longlong
0 iio_device_debug_attr_write_raw
0 iio_device_find_attr
0 iio_device_find_buffer_attr
0 iio_device_find_channel
0 iio_device_find_debug_attr
0 iio_device_get_buffer_attr
0 iio_device_get_buffer_attrs_count
0 iio_device_get_context
0 iio_device_get_data
0 iio_device_identify_filename
0 iio_device_set_data
0 iio_has_backend
0 iio_strerror
1 iio_buffer_destroy
1 iio_buffer_end
1 iio_buffer_push
1 iio_buffer_push_partial
1 iio_buffer_refill
1 iio_buffer_start
1 iio_channel_attr_get_filename
1 iio_channel_attr_read
1 iio_channel_attr_write
1 iio_channel_disable
1 iio_channel_enable
1 iio_channel_get_attrs_count
1 iio_channel_get_id
1 iio_channel_get_name
1 iio_channel_is_enabled
1 iio_channel_is_output
1 iio_channel_is_scan_element
1 iio_channel_read_raw
1 iio_channel_write_raw
1 iio_context_clone
1 iio_context_destroy
1 iio_context_get_attrs_count
1 iio_context_get_description
1 iio_context_get_devices_count
1 iio_context_get_name
1 iio_context_get_version
1 iio_context_get_xml
1 iio_context_info_get_description
1 iio_context_info_get_uri
1 iio_context_info_list_free
1 iio_context_set_timeout
1 iio_create_context_from_uri
1 iio_create_default_context
1 iio_create_local_context
1 iio_create_network_context
1 iio_create_scan_context
1 iio_create_xml_context
1 iio_device_attr_read
1 iio_device_attr_write
1 iio_device_create_buffer
1 iio_device_debug_attr_read
1 iio_device_debug_attr_write
1 iio_device_get_attrs_count
1 iio_device_get_channels_count
1 iio_device_get_debug_attrs_count
1 iio_device_get_id
1 iio_device_get_name
1 iio_device_get_sample_size
1 iio_device_get_trigger
1 iio_device_is_trigger
1 iio_device_reg_read
1 iio_device_reg_write
1 iio_device_set_kernel_buffers_count
1 iio_device_set_trigger
1 iio_get_backends_count
1 iio_library_get_version
1 iio_scan_context_destroy
1 iio_scan_context_get_info_list
2 iio_channel_get_attr
2 iio_channel_read
2 iio_channel_write
2 iio_context_get_attr
2 iio_context_get_device
2 iio_device_get_attr
2 iio_device_get_channel
2 iio_device_get_debug_attr
2 iio_get_backend
```
It should be straightforward to add things - once we decide which ones need to be added.

cristi-iacob, 2020-04-10:
I have made a document with these bindings and whether they are needed or not (with a short explanation for each of them). I have enabled the comments for this [document]( https://docs.google.com/spreadsheets/d/1IraOknY8DwshpZdgEbytlFMUN-Fs94yM0S6YIeNe8z8/edit#gid=0) (and disabled the edit mode - because this is a public page and anyone can change it in an unwanted way). The methods I am not sure about are marked with a question mark.

I will also start implementing the necessary methods and open a pull request with them.

rgetz, 2020-04-10:
Thanks for looking at this. Your doc looked pretty good.

My list was created by a little shell. Based on your list (getting rid of the C types), it would turn into:

```
for funct in $(grep ^__api iio.h | tr " " "\n" | grep "(" | sed 's/(.*$//' | grep -v -e _bool$ -e _double$ -e _longlong$ -e _all$ -e _raw$); 
  do echo "$(grep -c $funct ./bindings/python/iio.py) $funct" ;  
done | grep "^0"
```
after we are done, we should look at trying to automate things, so when someone adds a libiio function, we make sure a binding is added as well, or somehow mark certain functions as not necessary in a binding.

rgetz, 2020-04-11:
and if you are poking at the python bindings - it might be easier to fix the 15 [Codacy](https://app.codacy.com/gh/analogdevicesinc/libiio/dashboard?bid=17401592) issues pointed out for [bindings/python/iio.py]( https://app.codacy.com/gh/analogdevicesinc/libiio/file/43786369196/issues/source?bid=17401592&fileBranchId=17401592) first.

tfcollins, 2020-04-13:
I add a bunch of comments to the google doc.

rgetz, 2020-05-20:
I think this is all resolved - so closing.

closed 2020-05-20 by rgetz