analogdevicesinc/libiio #370: Running examples/dummy-iiostream.c Problem

samiemostafavi, 2020-02-10:
Hi,

I am trying to compile and run dummy-iiostream.c file but I get this error:

Library version: 0.18 (git tag: 8d6ad29)
* Acquiring IIO context
* Acquiring device my_dummy_device
* Initializing IIO streaming channels:
No scan elements found (make sure the driver built with 'CONFIG_IIO_SIMPLE_DUMMY_BUFFER=y')
* Destroying buffers
* Disassociate trigger
* Destroying context

Here is what I get when I run iio_info:

Library version: 0.18 (git tag: 8d6ad29)
Compiled with backends: local xml ip usb
IIO context created with local backend.
Backend version: 0.18 (git tag: 8d6ad29)
Backend description string: Linux mbp 5.0.0-32-generic #34~18.04.2-Ubuntu SMP Thu Oct 10 10:36:02 UTC 2019 x86_64
IIO context has 1 attributes:
	local,kernel: 5.0.0-32-generic
IIO context has 4 devices:
	iio:device0: als (buffer capable)
		2 channels found:
			intensity_both:  (input, index: 0, format: le:U32/32>>0)
			3 channel-specific attributes found:
				attr  0: hysteresis value: 0
				attr  1: input value: 0
				attr  2: sampling_frequency value: 5.000000
			illuminance:  (input)
			3 channel-specific attributes found:
				attr  0: hysteresis value: 0
				attr  1: input value: 0
				attr  2: sampling_frequency value: 5.000000
		2 buffer-specific attributes found:
				attr  0: data_available value: 0
				attr  1: watermark value: 1
		Current trigger: trigger0(als-dev0)
	iio:device1: my_dummy_device
		10 channels found:
			voltage0:  (input)
			3 channel-specific attributes found:
				attr  0: offset value: 7
				attr  1: raw value: 73
				attr  2: scale value: 0.001333
			activity_walking:  (input)
			1 channel-specific attributes found:
				attr  0: input value: 4
			accel_x:  (input)
			3 channel-specific attributes found:
				attr  0: calibbias value: -7
				attr  1: calibscale value: 0.000100
				attr  2: raw value: 34
			voltage1-voltage2:  (input)
			1 channel-specific attributes found:
				attr  0: raw value: 33
			voltage3-voltage4:  (input)
			1 channel-specific attributes found:
				attr  0: raw value: -34
			voltage0:  (output)
			1 channel-specific attributes found:
				attr  0: raw value: 0
			activity_running:  (input)
			1 channel-specific attributes found:
				attr  0: input value: 98
			steps:  (input)
			3 channel-specific attributes found:
				attr  0: calibheight value: 0
				attr  1: en value: 0
				attr  2: input value: 47
			sampling:  (input)
			1 channel-specific attributes found:
				attr  0: frequency value: 3.000000033
			voltage-voltage:  (input)
			1 channel-specific attributes found:
				attr  0: scale value: 0.000001344
		1 device-specific attributes found:
				attr  0: current_timestamp_clock value: realtime

	trigger0: als-dev0
		0 channels found:
	trigger1: instance1
		0 channels found:
		1 device-specific attributes found:
				attr  0: sampling_frequency value: 100

Thanks for the help!
Best,
Samie

rgetz, 2020-02-12:
Are you building on ARM or x86?

samiemostafavi, 2020-02-13:
I am building on x86.

rgetz, 2020-02-13:
Are you editing the kernel via Kconfig? (there are more options that are set in Kconfig) or just on the command line?

Last time I tried this it worked - but it was awhile ago.

samiemostafavi, 2020-02-13:
Its just on the command line.

rgetz, 2020-02-14:
try from `make menuconfig` - it's likely that other options will not be turned on if you do it from there.

It would be under `Device Drivers` -> `Industrial I/O support` (turn on ` Enable software IIO device support` and `Enable software triggers support`) then you can enable -> `IIO dummy driver`, and then you need to turn on `Event generation support` and `Buffered capture support`

If you want to do it from the command line it would be something all of these (and maybe more, there are things that Kconfig will turn on for you once you select those, that will not be done on the command line).:
```
CONFIG_IIO=y
CONFIG_IIO_BUFFER=y
CONFIG_IIO_KFIFO_BUF=y
CONFIG_IIO_CONFIGFS=y
CONFIG_IIO_TRIGGER=y
CONFIG_IIO_CONSUMERS_PER_TRIGGER=2
CONFIG_IIO_SW_DEVICE=y
CONFIG_IIO_DUMMY_EVGEN=y
CONFIG_IIO_SIMPLE_DUMMY=y
CONFIG_IIO_SIMPLE_DUMMY_EVENTS=y
CONFIG_IIO_SIMPLE_DUMMY_BUFFER=y
```

rgetz, 2020-02-22:
Ok - no updates in a week, and I think I provided the correct solution, so closing this.

If you still run into a problem - please re-open.

closed 2020-02-22 by rgetz