analogdevicesinc/libiio #373: incorrect channel type (in, out, in_out?)

rgetz, 2020-02-11:
I'm not sure if this is kernel issue or a libiio one.

```
iio_attr -a -c ad9361-phy out
Using auto-detected IIO context at URI "usb:3.20.5"
dev 'ad9361-phy', channel 'out' (input), attr 'voltage_filter_fir_en', value '0'
```

an input channel called `out`?

we have:
```
in_voltage_filter_fir_en
out_voltage_filter_fir_en
```

which follow the normal constructs for global voltage[n] attributes, but then we also have:

```
in_out_voltage_filter_fir_en
```

which I think effects both inputs and output channels, but might break the kernel ABI? or is libiio just not handling `in_out` channels?

?

mhennerich, 2020-02-11:
This convention is used for things that are both directions. It's only used once. It's not mainline and thus libiio doesn't handle it. I would have reverted it but I didn't want to break the API.
We have some workaround here:
https://github.com/analogdevicesinc/libad9361-iio/blob/master/ad9361_baseband_auto_rate.c#L64

rgetz, 2020-02-11:
OK - I will keep this open until we decide that the "in_out" (bidirectional attributes) are going upstream (and we should extend libiio) or if they aren't (and we should just leave it alone).

-Robin

rgetz, 2020-02-13:
Ok - now iio_info gives:

```
                        out:  (input, WARN:iio_channel_get_type()=UNKNOWN)
                        1 channel-specific attributes found:
                                attr  0: voltage_filter_fir_en value: 0
```

as a polite reminder...

rgetz, 2022-02-17:
closing - since this is a kernel driver issue.

closed 2022-02-17 by rgetz