analogdevicesinc/libiio #379: Input/output error (-5) when writing write only attributes using iio_attr

mbancisor, 2020-02-17:
dump_device_attributes() tries to do a read after each write (https://github.com/analogdevicesinc/libiio/blob/master/tests/iio_attr.c#L207). This fails for the write only attributes, returning:
ERROR: Input/output error (-5)

rgetz, 2020-02-17:
On the device:
```
--w-------    1 root     root          4096 Jan  1 00:00 ./iio:device1/multichip_sync
```

On a host:
```
iio_attr -a -d ad9361-phy  multichip_sync 0
Using auto-detected IIO context at URI "usb:3.8.5"
dev 'ad9361-phy', attr 'multichip_sync', value :ERROR: Permission denied (-13)
wrote 2 bytes to multichip_sync
dev 'ad9361-phy', attr 'multichip_sync', value :ERROR: Permission denied (-13)
```

That means the read didn't work, but the write did.

What exact device/attribute are you trying?

mhennerich, 2020-02-17:
ad9361-phy, multichip_sync doesn't implement a read method.

rgetz, 2020-02-17:
Agree - that is expected. A attribute which is write only - read will fail.

I'm saying that's OK.

dNechita, 2020-02-18:
Wouldn't it make sense to add an additional flag for cases where only a write is intended.
e.g.
`iio_attr -a -d ad9361-phy -w  multichip_sync 0`
, where -w means write-only.

rgetz, 2020-02-19:
I'm not sure. The only way that I'm aware to make sure the write worked as expected (or to see what value is actually written to an attribute), is to do a read afterwards.

That is what -q (quiet) does - is don't do the first read, don't print the status of the write, and only do the second one, for that specific purpose (to see what is actually there).

I don't see the harm that the error string provides.

dNechita, 2020-02-20:
@mbanciso is this behavior causing any inconveniences for you setup?

closed 2020-03-17 by mbancisor