analogdevicesinc/libiio #390: DNS SD doesn't work on Windows

rgetz, 2020-02-27:
As discussed in #388 - we don't support Win32 yet.

https://github.com/stammen/dnssd-uwp
has enough to make sure that Win10 would be supported (which I think is OK).

That would use these built in win10 features:
https://docs.microsoft.com/en-us/uwp/api/windows.networking.servicediscovery.dnssd

rgetz, 2020-03-29:
I made a pull request #412 - let me know if we should merge that as is, or if you think you will have time to look at adding windows 10 support before then.

rgetz, 2020-03-29:
I guess the steps would be to create a dns_sd_win10.c, this would include things like:

- a test for [win10](https://github.com/stammen/dnssd-uwp/blob/61e30060f1dc921e2fe4e4a7d95b15af95fbcf31/dnssd/WindowsVersionHelper.h#L48-L86) 
- dynamically load the dnssd-uwp DLL using LoadLibrary() & get pointers to the various [dnssd functions](https://github.com/stammen/dnssd-uwp/blob/61e30060f1dc921e2fe4e4a7d95b15af95fbcf31/DnssdClient/DnssdClient.cpp#L50-L89)
- Create a dnssd service watcher using the [above function](https://github.com/stammen/dnssd-uwp/blob/61e30060f1dc921e2fe4e4a7d95b15af95fbcf31/DnssdClient/main.cpp#L122-L127)
- Which calls your [callback](https://github.com/stammen/dnssd-uwp/blob/61e30060f1dc921e2fe4e4a7d95b15af95fbcf31/DnssdClient/main.cpp#L37-L69) which builds up the linked list, like the avahi and bonjour implementations do.
- free everything when done.

I think that should be it. All the above code is under a MIT license - which means we can include it into libiio, we just need to point back, and make sure to include the appropriate copyrights in it.

matejk, 2020-03-30:
Would it make sense to use an independent implementation like:

https://github.com/mjansson/mdns

rgetz, 2020-03-30:
> Would it make sense to use an independent implementation like:

I don't think so - as much as it pains me - I don't think adding another library is worth it.

rgetz, 2020-05-20:
This is being worked on as part of #508 , so closing this, as that (hopefully) will be merged soon.

closed 2020-05-27 by rgetz