analogdevicesinc/libiio #391: filtering issues with avahi/bonjour

rgetz, 2020-02-27:
As discussed in #388 we aren't doing something right in the avahi/bonjour code - since many devices (non-IIO) respond back when looking on the network.

rgetz, 2020-03-05:
Ok - I have things working, and I added it to the scan context, so now you get:

```
rgetz@brain:~/github/libiio/build$ ./tests/iio_info -s
Library version: 0.19 (git tag: 875d937)
Compiled with backends: local xml ip usb serial
Available contexts:
	0: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=104473222a87000618000600473ed57ae0 [usb:3.8.5]
	1: 192.168.1.120 Linux analog 4.14.0-g8a57ede #652 SMP PREEMPT Mon Feb 3 16:34:57 GMT 2020 armv7l [ip:analog.local]
	2: 192.168.2.1 (Analog Devices PlutoSDR Rev.B (Z7010-AD9364)), serial=104473222a87000618000600473ed57ae0 [ip:pluto.local]
```

however, this breaks `-a` in most applications:

```
rgetz@brain:~/github/libiio/build$ ./tests/iio_info -a
Library version: 0.19 (git tag: 875d937)
Compiled with backends: local xml ip usb serial
Multiple contexts found. Please select one using --uri:
	0: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=104473222a87000618000600473ed57ae0 [usb:3.8.5]
	1: 192.168.1.120 Linux analog 4.14.0-g8a57ede #652 SMP PREEMPT Mon Feb 3 16:34:57 GMT 2020 armv7l [ip:analog.local]
	2: 192.168.2.1 (Analog Devices PlutoSDR Rev.B (Z7010-AD9364)), serial=104473222a87000618000600473ed57ae0 [ip:pluto.local]
```
and
```
rgetz@brain:~/github/libiio/build$ ./tests/iio_attr -d -a
Multiple contexts found. Please select one using --uri:
	0: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=104473222a87000618000600473ed57ae0 [usb:3.8.5]
	1: 192.168.2.1 (Analog Devices PlutoSDR Rev.B (Z7010-AD9364)), serial=104473222a87000618000600473ed57ae0 [ip:pluto.local]
	2: 192.168.1.120 Linux analog 4.14.0-g8a57ede #652 SMP PREEMPT Mon Feb 3 16:34:57 GMT 2020 armv7l [ip:analog.local]
```

so I'm not sure if that is a good thing, or a bad one...

-Robin

rgetz, 2020-03-05:
This turns osc into this:
[image]

so it may need a tweak to handle the IP number at the front...

rgetz, 2020-03-05:
@tfcollins suggested that the list of devices might be better than the iio_context_description.

`ad7291+ad9361-phy+xadc+cf-ad9361-dds-core-lpc+cf-ad9361-lpc`
rather than
`Linux analog 4.14.0-g8a57ede #652 SMP PREEMPT Mon Feb 3 16:34:57 GMT 2020 armv7l`

Thoughts/opinions? (I think that is what I like too).

tfcollins, 2020-03-05:
Yes, 99% of the time I don't really care about the remote kernel version... I just want to know what the drivers are available.

rgetz, 2020-03-05:
```
rgetz@brain:~/github/libiio/build$ ./tests/iio_info -s
Library version: 0.19 (git tag: 875d937)
Compiled with backends: local xml ip usb serial
Available contexts:
	0: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=104473222a87000618000600473ed57ae0 [usb:3.8.5]
	1: 192.168.2.1 (Analog Devices PlutoSDR Rev.B (Z7010-AD9364)), serial=104473222a87000618000600473ed57ae0 [ip:pluto.local]
	2: 192.168.1.120 ad7291+ad9361-phy+xadc+cf-ad9361-dds-core-lpc+cf-ad9361-lpc [ip:analog.local]
```

better?

dNechita, 2020-03-06:
Can we have commas instead of plus sign? For me it's a little hard to follow the content of the line because + and - is alternating.

matejk, 2020-03-06:
I think that detection of IIOD network services using DNS-SD as it currently implemented, only gives the first device that was found, however there can be many of them on the network.

Shall this deficiency be resolved in the context of this issue also?

rgetz, 2020-03-06:
Yes - this is fixed at the same time.

I have a solution for Linux, but it I don't have a MAC to test `dns_sd_bonjour.c` 

now `ip:` still (randomly) attaches to the first one, but at least it finds all the others...
If you want to look at everything, then `iio_info -s` will print everything.

Handling this is better done in the applications side pretty easy too. I may change iio_attr to do that (if there are multiple scan contexts, find the one based on what you are looking for).

rgetz, 2020-03-06:
@dNechita 

```
Available contexts:
	0: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=104473222a87000618000600473ed57ae0 [usb:3.13.5]
	1: 192.168.1.120 (ad7291,ad9361-phy,xadc,cf-ad9361-dds-core-lpc,cf-ad9361-lpc) [ip:analog.local]
	2: 192.168.2.1 (Analog Devices PlutoSDR Rev.B (Z7010-AD9364)), serial=104473222a87000618000600473ed57ae0 [ip:pluto.local]
```

matejk, 2020-03-09:
I can improve code for Bonjour once the code for Avahi is ready.

rgetz, 2020-03-10:
Thanks - I think I need to shuffle some things around (move standard into network.c), and add a lock/mutex to the linked list - I can cause something things to go crazy when running this code while plugging some things into the network. Give me a few more days to track everything down.

rgetz, 2020-03-12:
@matejk - why don't you have a look now? 
https://github.com/analogdevicesinc/libiio/pull/395/files

Comments welcome.

Things like `iio_info` just work (no changes) to it.
```
rgetz@brain:~/github/libiio-dns/build$ ./tests/iio_info -s
Library version: 0.19 (git tag: 3583818)
Compiled with backends: local xml ip usb serial
Available contexts:
	0: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=104473222a87000618000600473ed57ae0 [usb:3.15.5]
	1: 192.168.1.120 (ad7291,ad9361-phy,xadc,cf-ad9361-dds-core-lpc,cf-ad9361-lpc) [ip:analog.local]

```

The goal (I think) would be to have a linked list that is shared between all the DNS SD implementations; which include three mandatory/common attributes, which everyone fills out: 
```
	char addr_str[DNS_SD_ADDRESS_STR_MAX];
	char *hostname;
	uint16_t port;
``` 
and other optional attributes for the specific implementations (avahi, bonjour, etc).

This way we can pass this to common functions in network.c, and have it manage things, and avoid some duplication in the specific implementations.

Does that work for you? If so - I can move some of the linked list processing into network.c, so you can re-use it.Then you are left with just filling out the structure...

?

-Robin

matejk, 2020-03-12:
Thanks, I'll take a look today.

matejk, 2020-03-12:
I'll see if I can make the resulting linked list without implementation specific data and hide all other stuff in avahi/bonjour code.

rgetz, 2020-03-12:
Thanks.

I'm also open to a new dns_sd.c file for common things, so network.c doesn't have as many ifdefs. That also might make things cleaner.

matejk, 2020-03-12:
I was also thinking about another dns sd file. I'll add it.

rgetz, 2020-03-13:
You made need this https://github.com/analogdevicesinc/libiio/pull/399/files from #399 to get things working properly.

matejk, 2020-03-14:
Code to perform discovery with CFNetSetServices is implemented on branch https://github.com/matejk/libiio/tree/gh-391-bonjour-discovery

Common code was moved to a new file dns_sd.c.

Changes are done on top of the PR #395 and #399.

rgetz, 2020-03-15:
Thanks alot.

If it is OK with you - I will let #399 get merged, and re-factor everything (#395 and your patches) into a few logical commits, and add your signed-off-by. Then that can get reviewed/merged before I look at the Windows 10 things that you pointed to before. (which should be straight forward now).

matejk, 2020-03-16:
Yes, of course I agree.

rgetz, 2020-03-28:
I was able to borrow a MAC from someone and test things out an a few different configurations (zero devices on network, single device on network, and multiple devices on network). I think this works now. I did need to tweak some of the mac implementation you did, but it was working for me.

Let me now if you see any issues.

matejk, 2020-03-30:
I added a few comments to the PR.

rgetz, 2020-03-30:
Ok - the final PR#412 is open, and I will close this. Thanks

PR #412 (2020-04-01): dns-sd (avahi/bonjour) improvements
This makes dns-sd work on Linux and MacOS for scan contexts. This still needs a Windows implementation. 

On MAC:
```
analogs-mini:build analog$ ./tests/iio_info -s
Available contexts:
	0: 192.168.86.35 (ad7291-ccbox,ad9361-phy,xadc,cf-ad9361-dds-core-lpc,adis16460,cf-ad9361-lpc) [ip:analog-2.local.]
```

on Linux (different network):
```
getz@brain:~/github/libiio$ ./build/tests/iio_info -s
Library version: 0.19 (git tag: d5b4247)
Compiled with backends: local xml ip usb serial
Available contexts:
	0: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)),  [... cut]
closed 2020-03-30 by rgetz