analogdevicesinc/libiio #398: all travis-ci builds on Linux give false positives

rgetz, 2020-03-12:
If you look at a recent build that fails, like:

https://travis-ci.org/github/analogdevicesinc/libiio/builds/661315711?utm_source=github_status&utm_medium=notification

The Linux builds are all passing - which is wrong. Looking at the logs:

/docker_build_dir/dns_sd_avahi.c:384: warning: implicit declaration of function 'avahi_simple_poll_free'
/docker_build_dir/dns_sd_avahi.c:384: error: 'struct dns_sd_discovery_data' has no member named 'poll'
make[2]: *** [CMakeFiles/iio.dir/dns_sd_avahi.c.o] Error 1
make[1]: *** [CMakeFiles/iio.dir/all] Error 2
make: *** [all] Error 2

But the script continues and provides a 0 error code on exit.

(I think) since the exit code is dropped from the make, since the script library runs and rate checks things (which functions).

commodo, 2020-03-12:
thanks for the find
splitting `make && make package` seems to have done the trick now

i'm still a bit puzzled about this;

i suspect it may have something to do with the return code;
`make` seems to have been returning `2` [instead of the usual `1` for error]
maybe in that case `make && make package` doesn't fail

i didn't dig too deep into this; and i probably won't
i'm happy with just splitting them;

rgetz, 2020-03-12:
Looks like it's fixed in #400 

it's wierd that that is the cause - I didn't think bash's `&&` operator did that...

```
rgetz@brain:~/github/libiio-dns$ cat /dev/null && echo $?
0
rgetz@brain:~/github/libiio-dns$ cat /dev/null1 && echo $?
cat: /dev/null1: No such file or directory
rgetz@brain:~/github/libiio-dns$ echo $? 
1
```

unless its some dash thing inside the script?

Thanks

PR #400 (2020-03-13): ci,build: split 'make && make package' commands
Fixes #398 

For some reason, doing `make && make package` doesn't seem to yield a
compiler/build error on Linux.
This may be due to how '/bin/sh' interprets command concatenation via '&&'.
Splitting the make commands seems to help.

See:
  https://travis-ci.org/github/analogdevicesinc/libiio/builds/661401114

This is essentially branch 'rgetz-review-zeroconfupdates' rebased with this
change.
closed 2020-03-12 by rgetz