analogdevicesinc/libiio #403: CMake build with -DWITH_LOCAL_CONFIG=ON fails

matejk, 2020-03-14:
I wanted to include support for reading local ini file.

```
$ cmake ../libiio -DWITH_LOCAL_CONFIG=ON
CMake Error at CMakeLists.txt:170 (message):
  WITH_LOCAL_CONFIG option requires libini to be installed
```

Build instruction to not contain information about the libini library.

How is this supposed to be built?

rgetz, 2020-03-14:
with libini

I will add it to the build instructions.

rgetz, 2020-03-14:
I just checked the pluto and m2k firmware - you need to check out https://github.com/pcercuei/libini/ and build/install it.

rgetz, 2020-03-14:
I updated the README_BUILD.md and https://wiki.analog.com/resources/tools-software/linux-software/libiio

rgetz, 2020-03-15:
I looked at adding this to the Linux images on the SD Cards:

```
diff --git a/adi_update_tools.sh b/adi_update_tools.sh
index 49b36d4..4e4da9c 100755
--- a/adi_update_tools.sh
+++ b/adi_update_tools.sh
@@ -1,4 +1,4 @@
-#!/bin/sh
+#!/bin/bash
 
 if [ "$(id -u)" != "0" ] ; then
        echo "This script must be run as root"
@@ -261,6 +262,24 @@ do
        # Non-essential applications, which help out sometime
        apt-get -y install gpsd gpsd-clients u-boot-tools evtest
 
+       # install libini
+       pushd $(pwd)
+       cd /usr/local/src
+        if [ -d "libini" ] ; then
+          cd ./libini
+          git pull
+        else
+         git clone https://github.com/pcercuei/libini.git
+          cd ./libini
+        fi
+       mkdir -p build
+       cd build
+       rm * -rf
+       cmake ../
+       make
+       make install
+
        if [ ! -f /etc/fw_env.config ]
        then
          cp ./fw_env.config /etc/
```

but this fails, since Paul has a min cmake of 3.8, and in the 2019 release, it is still cmake 2.12

libini needs this change:
```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 0a7dee6..c3773cd 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -1,5 +1,9 @@
-cmake_minimum_required(VERSION 3.1.0)
-project(libini LANGUAGES C VERSION 0.1)
+cmake_minimum_required(VERSION 2.8.7)
+project(libini C)
+
+set(LIBINI_VERSION_MAJOR 0)
+set(LIBINI_VERSION_MINOR 1)
+set(VERSION "${LIBINI_VERSION_MAJOR}.${LIBINI_VERSION_MINOR}")
 
 include(GNUInstallDirs)
 
@@ -24,8 +28,8 @@ add_definitions(-DLIBINI_EXPORTS=1)
 
 add_library(ini SHARED ${LIBINI_CFILES} ${LIBINI_HEADERS})
 set_target_properties(ini PROPERTIES
-       VERSION ${PROJECT_VERSION}
-       SOVERSION ${PROJECT_VERSION_MAJOR}
+       VERSION ${VERSION}
+       SOVERSION ${LIBINI_VERSION_MAJOR}
        PUBLIC_HEADER ${LIBINI_HEADERS}
 )
```

I'm not sure if @mhennerich pr @pcercuei wants to fork libini into the analogdevicesinc organization, and maintain/make those change locally? or push things back to pauls 's project?

Once those are turned on, then `-DWITH_LOCAL_CONFIG=ON` would be turned on when building libiio, and a /etc/libiio.ini set up based on:
```
root@analog:/usr/local/src/libini# cat /sys/firmware/devicetree/base/model
Xilinx Zynq ZED
root@analog:/usr/local/src/libini# fru-dump -b -i /sys/devices/soc0/fpga-axi@0/41620000.i2c/i2c-1/1-0050/eeprom | grep "Part Number"
Part Number     : AD-FMCOMMS2-EBZ
root@analog:/usr/local/src/libini# fru-dump -b -i /sys/devices/soc0/fpga-axi@0/41620000.i2c/i2c-1/1-0050/eeprom | grep "Serial Number"
Serial Number   : 00100
root@analog:/usr/local/src/libini# fru-dump -b -i /sys/devices/soc0/fpga-axi@0/41620000.i2c/i2c-1/1-0050/eeprom | grep "Manu"
Manufacturer    : Analog Devices
```

That should give us similar to things that are on Pluto/M2k's `/etc/libiio.ini`
```
# cat /etc/libiio.ini 
[Context Attributes]
hw_model=Analog Devices PlutoSDR Rev.B (Z7010-AD9364)
hw_serial=104473222a87000618000600473ed57ae0
```

```
# cat /etc/libiio.ini 
[Context Attributes]
hw_model=Analog Devices AD-FMCOMMS2-EBZ (on Xilinx Zynq ZED)
hw_serial=00100
```

and make everything show up nicer on a network scan...

Thanks
-Robin

matejk, 2020-03-16:
Having some global IIO attributes in a ini file is very useful.

IMO this should be documented somewhere (how to use it, standard hw_attributes...).

PR #404 (2020-03-16): README_BUILD.md : update with a few more Prerequisites
Fix spelling and resolve #403 

Signed-off-by : Robin Getz <robin.getz@analog.com>
commit b9acd576 (2020-03-16): README_BUILD.md : update with a few more Prerequisites

Fix spelling and resolve #403 

Signed-off-by : Robin Getz <robin.getz@analog.com>
closed 2020-03-16 by rgetz