analogdevicesinc/libiio #425: MACRO redefintion on windows build

reid-p, 2020-04-07:
Just started getting a warning from MSVC build of libiio that the ERROR macro is redefined.

This started with the "coverity: fix contex.c Untrusted value as argument" commit.
Specifically debug.h defines an ERROR macro, and now that context.c include "network.h" which 
include the windows winsock headers which for me are include wingdi.h which also define an ERROR macro things don't build.

windgi.h:
#define ERROR               0

debug.h
#  define ERROR(...) \
    fprintf(stderr, "ERROR: " __VA_ARGS__)

Thoughts?

pcercuei, 2020-04-07:
I use the same debug.h in other projects, and at some point I had to rename the macros as I was facing the same problem. Names like 'ERROR' or 'DEBUG' are too common for macros. I renamed them to pr_debug, pr_info, pr_warn and pr_err.

rgetz, 2020-04-07:
Ok - I will do the same here...

rgetz, 2020-04-07:
Should be fixed when #427 is committed to master. If you have any more issues - please let us know, and re-open.

commit 135c1892 (2020-04-08): Merge pull request #427 from analogdevicesinc/rgetz-fix-name-space-collisions

fix #425: add IIO_ prefix to INFO, WARN, DEBUG and ERROR macros
commit 8188c16b (2020-04-08): fix #425: add IIO_ prefix to INFO, WARN, DEBUG and ERROR macros

this namespacing will help avoid collisions like #425 on Windows
and other platforms, that libiio is compiled on.
closed 2020-04-08 by dNechita