analogdevicesinc/libiio #450: Getting "Unable to allocate buffer: Invalid argument" when trying to use iio_readdev

Panky-codes, 2020-04-22:
Hello. 
I want to use libiio for my project so I was just playing around cli tools that you have built to see how libiio works. 
This is my output when I execute iio_info:
```
root@beaglebone:/sys/bus/iio/devices/iio:device0# /home/debian/bin/iio_info
Library version: 0.14 (git tag: v0.14)
Compiled with backends: local xml ip usb serial
IIO context created with local backend.
Backend version: 0.14 (git tag: v0.14)
Backend description string: Linux beaglebone 4.14.108-ti-r113 #1 SMP PREEMPT Wed Jul 31 00:01:10 UTC 2019 armv7l
IIO context has 1 attributes:
        local,kernel: 4.14.108-ti-r113
IIO context has 1 devices:
        iio:device0: 44e0d000.tscadc:adc.0.auto (buffer capable)
                8 channels found:
                        voltage0:  (input, index: 0, format: le:u12/16>>0)
                        1 channel-specific attributes found:
                                attr  0: raw value: 1290
                        voltage1:  (input, index: 1, format: le:u12/16>>0)
                        1 channel-specific attributes found:
                                attr  0: raw value: 2357
                        voltage2:  (input, index: 2, format: le:u12/16>>0)
                        1 channel-specific attributes found:
                                attr  0: raw value: 2925
                        voltage3:  (input, index: 3, format: le:u12/16>>0)
                        1 channel-specific attributes found:
                                attr  0: raw value: 952
                        voltage4:  (input, index: 4, format: le:u12/16>>0)
                        1 channel-specific attributes found:
                                attr  0: raw value: 1108
                        voltage5:  (input, index: 5, format: le:u12/16>>0)
                        1 channel-specific attributes found:
                                attr  0: raw value: 1817
                        voltage6:  (input, index: 6, format: le:u12/16>>0)
                        1 channel-specific attributes found:
                                attr  0: raw value: 1899
                        voltage7:  (input, index: 7, format: le:u12/16>>0)
                        1 channel-specific attributes found:
                                attr  0: raw value: 3832
                1 buffer-specific attributes found:
                                attr  0: watermark value: 1

```
This looks good to me as I have a voltage source connected to the `voltage0` channel (1290 corresponds correctly to 0.56V).  
But when I run iio_readdev, I am getting an error as follows:
```
root@beaglebone:/sys/bus/iio/devices/iio:device0# /home/debian/bin/iio_readdev iio:device0 voltage0
Unable to allocate buffer: Invalid argument
root@beaglebone:/sys/bus/iio/devices/iio:device0# /home/debian/bin/iio_readdev -b 100 iio:device0 voltage0
Unable to allocate buffer: Invalid argument
```
Could someone tell what I am missing here to make it work? Any help would be highly appreciated. 
Let me know if you need more logs.

rgetz, 2020-04-22:
`iio_readdev` is for reading high speed buffers,not for low speed interfaces.

there are some example on how to find the high speed interfaces at:
https://wiki.analog.com/resources/tools-software/linux-software/libiio/iio_readdev

rgetz, 2020-04-22:
And, version 0.14 is pretty old (Jan 29, 2018) you might want to upgrade.

Panky-codes, 2020-04-22:
Regarding your first comment,
Thanks for the information. I think that wiki is useful. 
Regarding your second comment,
It is funny because I downloaded this tar file libiio-0.19.g5f5af2e-raspbian-9-armhf.tar.gz from your release page. Either it is a printing bug or someone wrongly packaged it. 
```
armlibiio/usr : ls -la lib/arm-linux-gnueabihf/libiio.so.0 
383108 Feb 14 12:21 lib/arm-linux-gnueabihf/libiio.so.0.19
```
The date the library was packaged seems to agree with the release date of 0.19.

rgetz, 2020-04-22:
The builds are all autogenerated - I will check them out. For now, check out:
https://wiki.analog.com/resources/tools-software/linux-software/libiio#understanding_which_libiio_is_being_used

Panky-codes, 2020-04-23:
Thanks for your reply. And it was my mistake regarding the version. It looks like beaglebone comes inbuilt with libiio and my binary was linking to the existing library in the /usr/lib folder. I was not aware of that. 
Thanks for your help. I will close the issue now.

closed 2020-04-23 by Panky-codes

rgetz, 2020-04-23:
Glad we could help