analogdevicesinc/libiio #474: codebase isn't c99 compliant? (double underscore reserved words?)

rgetz, 2020-04-29:
I noticed when turning on more warnings on clang, that C99 states:

-----
## 7.1.3 Reserved identifiers
Each header declares or defines all identifiers listed in its associated subclause, and optionally declares or defines identifiers listed in its associated future library directions subclause and identifiers which are always reserved either for any use or for use as file scope identifiers.
- All identifiers that begin with an underscore and either an uppercase letter or another underscore are always reserved for any use.
- All identifiers that begin with an underscore are always reserved for use as identifiers with file scope in both the ordinary and tag name spaces.
------

The [libc doc](https://www.gnu.org/software/libc/manual/html_node/Reserved-Names.html) says similar things:

----
## 1.3.3 Reserved Names

reserved names include all external identifiers (global functions and variables) that begin with an underscore (‘_’) and all identifiers regardless of use that begin with either two underscores or an underscore followed by a capital letter are reserved names. This is so that the library and header files can define functions, variables, and macros for internal purposes without risk of conflict with names in user programs.

----

We could read that as those are instructions for applications, and don't apply to libraries, like libiio, but I'm not sure C99 was written like that...

We use double underscore alot (__api, etc) in iio.h

More info can be found [here](https://wiki.sei.cmu.edu/confluence/display/c/DCL37-C.+Do+not+declare+or+define+a+reserved+identifier)

It's not hard to change (a couple minutes of sed), but it's alot of churn...

```
 bindings/matlab/iio-wrapper.h |   12 -
 dns_sd_avahi.c                |    6 
 examples/ad9361-iiostream.c   |    2 
 examples/ad9371-iiostream.c   |    4 
 examples/adrv9009-iiostream.c |    4 
 examples/dummy-iiostream.c    |    4 
 iio-private.h                 |   18 +-
 iio.h                         |  314 ++++++++++++++++++++----------------------
 libini/ini.h                  |   24 +--
 man/make_man.sh               |    2 
 10 files changed, 197 insertions(+), 193 deletions(-)
```

tfcollins, 2020-05-01:
This seems kinda nitpicky to me. IMHO if we get a name clash fix it then.

dNechita, 2020-05-05:
> This seems kinda nitpicky to me. IMHO if we get a name clash fix it then.

I'm afraid that will never get a name collision but an undefined behavior instead. This guy https://www.youtube.com/watch?v=0vJXQ2VOzEs explains this much better. I know it' about C++ but I think the concept is the same.

rgetz, 2020-05-05:
"undefined behavior" in that context is a name clash. 

the compiler should throw an error about things being redefined.

rgetz, 2020-05-12:
I think I will close as "Won't fix" as not many people care (until there is a problem).

closed 2020-05-12 by rgetz