analogdevicesinc/libiio #478: determine buffer over/under-flow in general or with iio_readdev / iio_writedev

hayguen, 2020-05-03:
is there a way to determine a buffer overflow in the C API?
if possible, can someone add it to iio_readdev?

took me a while to recognize, that the default buffer size is too small to receive a 2.1 Msps signal, which is quite near the minimum samplerate.
what would be a good rule for buffer size?
in my case 50ms (samplerate/20) looked to be a good value, but smaller values might also do ..

similar for buffer underflow and iio_writedev.

tfcollins, 2020-05-03:
This is basically what iio_adi_xflow_check does. Have you tried that?

-Travis

hayguen, 2020-05-03:
no, i've had not seen that and i've not tried for now.
the additional monitoring thread contains some unobvious magic like iio_device_reg_read() and some undocumented magic numbers inside ..

ideally, i would like to see that additional monitoring thread inside iio_readdev and writedev,
that i don't need an extra program/tool ..
it should be inside iio_readdev and writedev that i can be sure, that everything is fine, when running in dynamic high CPU load environment, e.g. a slower Raspberry Pi model

tfcollins, 2020-05-04:
These magic numbers are specific to the DMA engines we maintain, so iio_adi_xflow_check is really only specific to certain platforms. Primarily ones with FPGAs that use our HDL. So having a universal underflow/overflow check is not totally possible right now.

Are you using iio_readdev and iio_writedev to actually pipe data into your applications? Would you expect overflow/underflow to be exported through stderr or something else? What platforms are you using?

-Travis

mhennerich, 2020-05-04:
> what would be a good rule for buffer size?
There is no golden rule. It's always a trade-off between overhead and latency.
Large buffers are always good for performance but bad for latency.

hayguen, 2020-05-04:
> These magic numbers are specific to the DMA engines we maintain, so iio_adi_xflow_check is really only specific to certain platforms. Primarily ones with FPGAs that use our HDL. So having a universal underflow/overflow check is not totally possible right now.

certain platforms: you mean the AD/DA platforms like Adalm Pluto or something else?
can you list a few models, on which buffer over/under-flows could be detected?

> Are you using iio_readdev and iio_writedev to actually pipe data into your applications? Would you expect overflow/underflow to be exported through stderr or something else?

yes, i want to pipe data from/to 3rd party applications or my own application. if the parts look good, i'll just glue the different apps/tools together in a bash or python script.
for manual tests from a shell/command prompt, i'd also like to stream the binary data from/into a file .. and have a look inside afterwards.
error report on stderr would be very useful and sufficient - in both of above usages.

in addition to an error report, i'd also suggest to optionally abort streaming, e.g. when a number of errors is reached. the error report could be silenced after having reached some watermark, e.g. no more reports after having reported 10 times.

> What platforms are you using?

currently the Raspberry Pi models 3B+ and 4B. for laterer i'll see/test, if i can get it run on older models like 2B and 3B. if i see, that performance is insufficient, i'll look for small/mobile x64 models .. with Linux or Windows.

hayati

tfcollins, 2020-05-04:
> certain platforms: you mean the AD/DA platforms like Adalm Pluto or something else?
> can you list a few models, on which buffer over/under-flows could be detected?

This would be anything that uses our DMA and associated IIO driver. Anything in this table (at least) https://wiki.analog.com/resources/tools-software/linux-software/embedded_arm_images?s[]=zynq&s[]=sd#preparing_the_image + Pluto and M2K.

We support many devices (900+) that use these tools as well, so those flags would be very niche in the scheme of things.

Thanks for the feedback on what you would like to see. I can't promise anything but we'll try to take this into account when we make updates and enhancements. We do take PRs as well :)

-Travis

rgetz, 2020-05-12:
Since there isn't a standard Linux DMA API to report Overflows/UnderFlows (most DMA transactions are packet based, and handle backpressure, think Ethernet, USB, PCIe, SATA, etc) there isn't a good way to do this on most generic systems. 

I'm closing this - if you need more info - please feel free to re-open.

closed 2020-05-12 by rgetz