analogdevicesinc/libiio #485: iio.py needs a little more work...

rgetz, 2020-05-07:
The current iio.py has a few more issues.

[image]

- clean up the todo
```
bindings/python/iio.py:1155:2: W0511: TODO(pcercuei): Use a dictionary for the channels. (fixme)
bindings/python/iio.py:1383:2: W0511: TODO(pcercuei): Use a dictionary for the devices. (fixme)
```
- in some classes. they have 0 public methods:
```
bindings/python/iio.py:85:0: R0903: Too few public methods (0/2) (too-few-public-methods)
bindings/python/iio.py:89:0: R0903: Too few public methods (0/2) (too-few-public-methods)
bindings/python/iio.py:93:0: R0903: Too few public methods (0/2) (too-few-public-methods)
bindings/python/iio.py:97:0: R0903: Too few public methods (0/2) (too-few-public-methods)
bindings/python/iio.py:101:0: R0903: Too few public methods (0/2) (too-few-public-methods)
bindings/python/iio.py:105:0: R0903: Too few public methods (0/2) (too-few-public-methods)
bindings/python/iio.py:109:0: R0903: Too few public methods (0/2) (too-few-public-methods)
bindings/python/iio.py:703:0: R0903: Too few public methods (1/2) (too-few-public-methods)
bindings/python/iio.py:734:0: R0903: Too few public methods (1/2) (too-few-public-methods)
bindings/python/iio.py:763:0: R0903: Too few public methods (1/2) (too-few-public-methods)
```
- need to be implented in a metaclass
```
bindings/python/iio.py:696:21: E1101: Instance of '_attr' has no '_read' member (no-member)
bindings/python/iio.py:697:24: E1101: Instance of '_attr' has no '_write' member (no-member)
```
- code cleanups
```
bindings/python/iio.py:1437:40: R1714: Consider merging these comparisons with "in" to 'name_or_id in (x.name, x.id)' (consider-using-in)
```

rgetz, 2020-05-13:
There are 19 issues left - most might just be ignored...

Have a look later.

tfcollins, 2020-05-13:
Almost there :)

rgetz, 2020-05-18:
With #501, we are down to:
```
************* Module iio
/home/rgetz/github/libiio/bindings/python/iio.py:1167:2: W0511: TODO(pcercuei): Use a dictionary for the channels. (fixme)
/home/rgetz/github/libiio/bindings/python/iio.py:1395:2: W0511: TODO(pcercuei): Use a dictionary for the devices. (fixme)
/home/rgetz/github/libiio/bindings/python/iio.py:779:4: W0235: Useless super delegation in method '__init__' (useless-super-delegation)
/home/rgetz/github/libiio/bindings/python/iio.py:806:4: W0235: Useless super delegation in method '__init__' (useless-super-delegation)
/home/rgetz/github/libiio/bindings/python/iio.py:1288:4: W0235: Useless super delegation in method '__init__' (useless-super-delegation)
```

I think W0235 should be turned on for the examples, but need to be turned off for iio.py; (since we still want compatibility with Python 2 - correct?

rgetz, 2020-05-20:
I think we are OK now, so closing.

closed 2020-05-20 by rgetz