analogdevicesinc/libiio #517: Build on Mac with WITH_LOCAL_BACKEND

pooriayaghini, 2020-05-23:
Hi, 

I am trying to build the Libiio for Mac with WITH_LOCAL_BACKEND switch but the eventfd.h library used, does not exist on Mac. I am wondering if this option is supported on Mac?

**Build Command:**
cmake ../ -DPYTHON_BINDINGS=ON -DWITH_EXAMPLES=ON -DWITH_LOCAL_BACKEND=ON -DWITH_NETWORK_BACKEND=ON

**Error:**
/libiio/local.c:35:10: fatal error: 'sys/eventfd.h' file not found

Thanks,
PY

tfcollins, 2020-05-23:
Why do you need the local backend on macOS? I don't think the macOS kernel supports IIO. You should only need the remotes.

-Travis

rgetz, 2020-05-26:
yeah, as Travis states - `local` means IIO devices are attached to local SPI, I2C and exposed in sysfs (The Linux sysfs) Since IIO isn't supported in the MAC kernel - local doesn't make sense.

If you need anything else - please repoen.

closed 2020-05-26 by rgetz