analogdevicesinc/libiio #520: msvc build warning

reid-p, 2020-05-28:
The recent commit "Win: move to safe version of sscanf" is resulting in a build warning for me.
Specifically

`xml.c(153,25): error C2220: the following warning is treated as an error
xml.c(153,25): warning C4477: 'sscanf_s' : format string '%c' requires an argument of type 'unsigned int', but variadic argument 2 has type 'size_t'
xml.c(153,25): message : this argument is used as a buffer size
xml.c(153,25): warning C4477: 'sscanf_s' : format string '%c' requires an argument of type 'unsigned int', but variadic argument 4 has type 'size_t'
xml.c(153,25): message : this argument is used as a buffer size
xml.c(165,25): warning C4477: 'sscanf_s' : format string '%c' requires an argument of type 'unsigned int', but variadic argument 2 has type 'size_t'
xml.c(165,25): message : this argument is used as a buffer size
xml.c(165,25): warning C4477: 'sscanf_s' : format string '%c' requires an argument of type 'unsigned int', but variadic argument 4 has type 'size_t'
xml.c(165,25): message : this argument is used as a buffer size`

Casting the sizeof params to int solves the problem. ie:
`					&e, (unsigned int)sizeof(e), &s, (unsigned int)sizeof(s),`

tfcollins, 2020-05-28:
Can you try a more recent commit? I believe this was already fixed here https://github.com/analogdevicesinc/libiio/commit/282c10dd9b1de6ef4a1ac975965cff89444bcb04

reid-p, 2020-05-28:
On at commit : 831d858  Merge pull request #515 from analogdevicesinc/rgetz-coverity-fix-resource-leak

I'm doing a 64 bit build if that helps.

tfcollins, 2020-05-28:
Slightly confused on your last reply. Are you using 831d858? and what compiler are you using?

-Travis

rgetz, 2020-05-28:
yeah, we compile without warnings on master tip (32-bit and 64) on CI, or it would error.

Check Win Cmake flags at:

https://github.com/analogdevicesinc/libiio/blob/master/CMakeLists.txt#L62-L65

and 

https://github.com/analogdevicesinc/libiio/blob/master/CMakeLists.txt#L161-L164

log at:
https://ci.appveyor.com/project/analogdevicesinc/libiio/branch/master
```
-- The C compiler identification is MSVC 18.0.40629.0
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio 12.0/VC/bin/cl.exe

<snip>

-- The C compiler identification is MSVC 18.0.40629.0
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio 12.0/VC/bin/x86_amd64/cl.exe
```

reid-p, 2020-05-28:
Sorry, Yes using everything up to commit 831d858.
Compiler is:
Microsoft (R) C/C++ Optimizing Compiler Version 19.25.28614 for x64

rgetz, 2020-05-29:
Ok - I can replicate things on AppVeyor - and will merge fixes when everything is resolved. Thanks for the bug report. 

Just as a question - what version of MSVC does it make sense to release binaries on? oldest? or latest? I think today its 2015

rgetz, 2020-05-31:
Ok - this will be fixed on #523 - if you still see issues - let us know. Closing this for now.

closed 2020-05-31 by rgetz

reid-p, 2020-06-01:
> 
> Just as a question - what version of MSVC does it make sense to release binaries on? oldest? or latest? I think today its 2015

Tested the fix and all good. thanks.
I'm tend to use fairly recent MSVC, 2019 (15.5), to get the latest c++ support so I'd suggest latest.