analogdevicesinc/libiio #524: Update infrastructure on Appveyor

rgetz, 2020-05-31:
We are still using MSVS 2015, and 2019 is available - so we should be using that. 

This will also update the INNO installer from v5 to v6, which will require some changes (some installer languages got demoted to Unofficial, which aren't installed - we need do download them by hand).

We should also update the way that we are handling the windows runtime, and just have our installer download it. Similar to what is described here: https://codereview.stackexchange.com/questions/222996/installing-the-vc-redistributable-201520172019-for-both-x86-and-x64-using-in

The version of libusb we are using on the Windows builds is from 2015, we should update, and use official releases if possible. Same for libxml2 and libserialport (there have been alot of Windows bug fixes for [libserialport](https://sigrok.org/gitweb/?p=libserialport.git;a=log;h=HEAD) ) the official Windows version of libxml hasn't been updated since 2015 (as far as I can tell),

rgetz, 2020-06-01:
When updating inno installer - best thing to do is download:
https://downgit.github.io/#/home?url=https://github.com/jrsoftware/issrc/tree/master/Files/Languages

and use all those languages in:
https://github.com/analogdevicesinc/libiio/blob/master/libiio.iss.cmakein#L18-L45

rgetz, 2020-06-01:
Doc on VisualStudio Generators is at:

https://cmake.org/cmake/help/git-stage/generator/Visual%20Studio%2014%202015.html https://cmake.org/cmake/help/git-stage/generator/Visual%20Studio%2015%202017.html

For Cmake > 3.1, use:
```
cmake -G "Visual Studio 14 2015" -A Win32
cmake -G "Visual Studio 15 2017" -A Win32
cmake -G "Visual Studio 16 2019" -A Win32

cmake -G "Visual Studio 14 2015" -A x64
cmake -G "Visual Studio 15 2017" -A x64
cmake -G "Visual Studio 16 2019" -A x64
```

Since libusb and libxml aren't built for ARM or ARM64, don't worry about those.

rgetz, 2020-06-02:
Might also want to look at moving from AppVeyor to Azure DevOps.

rgetz, 2020-06-11:
Need to make sure -Werror is set _inside_ the docker or qemu images.

pcercuei, 2021-05-17:
We moved away from AppVeyor to Azure - is this bug report still valid?

rgetz, 2022-02-17:
Nope - closing.