analogdevicesinc/libiio #525: Wish: Conda package for libiio

petrushy, 2020-05-31:
Hi,

See also closed issue on pyadi-iio: https://github.com/analogdevicesinc/pyadi-iio/issues/92

It would be highly beneficial for less technically oriented to have a conda package for libiio (primary the python interface) that can be install in a relocatable non-sudo environment.

Regards!

petrushy, 2020-05-31:
Ah got so exited that I started on this. Got something that I think is working for linux only but very untested. Now at least I can get this running in anaconda python. :)

I would be happy if you would be interested in taking this to something more than a hack and a proper package on conda-forge. :)

https://github.com/petrushy/conda-recipes

see libiio and pyadi-iio folders.

petrushy, 2020-06-01:
Oh seems like someone already did it this weekend!
https://anaconda.org/ryanvolz/libiio/files

Thanks @ryanvolz!

ryanvolz, 2020-06-01:
@petrushy is right, I actually am a good way toward getting a package for `libiio` onto conda-forge. See https://github.com/conda-forge/staged-recipes/pull/11688. I haven't done the python bindings yet, but that can come next!

As an aside, the reason I'm working on this package is that me and my colleagues at MIT Haystack are running an Open Radar workshop right now (literally right now) and we wanted a good way to get the participants up and running without lots of hand holding. We're using ADALM-PLUTOs, so of course we needed `libiio`. I'm sure we'll learn things from how this goes to make the packages better.

rgetz, 2020-06-26:
If you have any comments - for improvements - let us know.

Otherwise I will close this next week.

petrushy, 2020-06-26:
@ryanvolz, I think you have a solution now to the conflict between these packages? As you have made a conda-forge package (https://anaconda.org/conda-forge/libiio) I propose we close this issue.

petrushy, 2020-06-26:
And again, thanks to both of you of the work...

closed 2020-06-28 by rgetz