analogdevicesinc/libiio #547: Sphinx errors when building 0.20

nullr0ute, 2020-06-06:
Getting the following sphinx errors when building the 0.20 release.

<pre>
Running Sphinx v2.2.2
making output directory... done
building [mo]: targets for 0 po files that are out of date
building [html]: targets for 11 source files that are out of date
updating environment: [new config] 11 added, 0 changed, 0 removed
reading sources... [  9%] buffer
reading sources... [ 18%] channel
reading sources... [ 27%] context
reading sources... [ 36%] device
reading sources... [ 45%] examples
reading sources... [ 54%] iio_attr
reading sources... [ 63%] iio_info
reading sources... [ 72%] iio_readdev
reading sources... [ 81%] iio_writedev
reading sources... [ 90%] index
reading sources... [100%] trigger

WARNING: autodoc: failed to import class 'Buffer' from module 'iio'; the following exception was raised:
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/sphinx/ext/autodoc/importer.py", line 32, in import_module
    return importlib.import_module(modname)
  File "/usr/lib64/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/perobins/fedora/packages/libiio/libiio-0.20/bindings/python/iio.py", line 236, in <module>
    _get_backends_count = _lib.iio_get_backends_count
  File "/usr/lib64/python3.8/ctypes/__init__.py", line 386, in __getattr__
    func = self.__getitem__(name)
  File "/usr/lib64/python3.8/ctypes/__init__.py", line 391, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: /usr/bin/python3: undefined symbol: iio_get_backends_count

WARNING: autodoc: failed to import class 'Channel' from module 'iio'; the following exception was raised:
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/sphinx/ext/autodoc/importer.py", line 32, in import_module
    return importlib.import_module(modname)
  File "/usr/lib64/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/perobins/fedora/packages/libiio/libiio-0.20/bindings/python/iio.py", line 236, in <module>
    _get_backends_count = _lib.iio_get_backends_count
  File "/usr/lib64/python3.8/ctypes/__init__.py", line 386, in __getattr__
    func = self.__getitem__(name)
  File "/usr/lib64/python3.8/ctypes/__init__.py", line 391, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: /usr/bin/python3: undefined symbol: iio_get_backends_count

WARNING: autodoc: failed to import class 'DataFormat' from module 'iio'; the following exception was raised:
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/sphinx/ext/autodoc/importer.py", line 32, in import_module
    return importlib.import_module(modname)
  File "/usr/lib64/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
[... 275 lines cut ...]
    func = self.__getitem__(name)
  File "/usr/lib64/python3.8/ctypes/__init__.py", line 391, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: /usr/bin/python3: undefined symbol: iio_get_backends_count

WARNING: autodoc: failed to import class 'Trigger' from module 'iio'; the following exception was raised:
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/sphinx/ext/autodoc/importer.py", line 32, in import_module
    return importlib.import_module(modname)
  File "/usr/lib64/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/perobins/fedora/packages/libiio/libiio-0.20/bindings/python/iio.py", line 236, in <module>
    _get_backends_count = _lib.iio_get_backends_count
  File "/usr/lib64/python3.8/ctypes/__init__.py", line 386, in __getattr__
    func = self.__getitem__(name)
  File "/usr/lib64/python3.8/ctypes/__init__.py", line 391, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: /usr/bin/python3: undefined symbol: iio_get_backends_count
</pre>

rgetz, 2020-06-06:
We build all the time with doc on CI, and don’t see this issue. I think all the CI use python 3.7 - I will set up something, and see if I can replicate. 

Thanks for reporting.

nullr0ute, 2020-06-06:
This is Fedora 32 with python 3.8, sphinx 2.2.11, sphinx_rtd_theme 0.4.3. I can also test against python 3.9 pre release if that's useful.

rgetz, 2020-06-06:
the "best" help would be figure out what's up with our CI, so you don't run into that again :)

We test on Centos 6, 7, 8.

Install all the packages:
https://github.com/analogdevicesinc/libiio/blob/master/CI/travis/before_install_linux

Build everything:
https://github.com/analogdevicesinc/libiio/blob/master/CI/travis/make_linux

Here is the build log for Centos 8 (with python 3.6.8 & Sphinx v3.0.4)
https://travis-ci.org/github/analogdevicesinc/libiio/jobs/695276801#L2649

I don't grok all the Fedora, CentOS, and RHEL interrelations - but we thought that was OK from a coverage standpoint - looks like not...

Looks like the other systems we build doc on is limited to xenial (Python 2.7.12 & Sphinx v3.0.4)
https://travis-ci.org/github/analogdevicesinc/libiio/jobs/695276797

I can fix that, to make sure we build doc on recent versions of everything (including darwin), but I still don't know if that would have caught everything.

Maybe - the issue is - the way that [we build things](https://github.com/analogdevicesinc/libiio/blob/master/CI/travis/make_linux). the python bindings need to see the library installed before it's built. We build without doc, without python, and install things, so when we build with python, and doc, everything is found properly on path...

-Robin

nullr0ute, 2020-06-07:
> I don't grok all the Fedora, CentOS, and RHEL interrelations - but we thought that was OK from a coverage standpoint - looks like not...

Fedora is the upstream of RHEL, CentOS is basically a community clone of RHEL (effectively downstream). Fedora has all the leading edge stuff so things like the latest gcc 10, latest python etc. It also often has newer, and as a result often tighter, compiler flags which can show up other issues.

> Maybe - the issue is - the way that [we build things](https://github.com/analogdevicesinc/libiio/blob/master/CI/travis/make_linux). the python bindings need to see the library installed before it's [...]

We also build everything in a chroot so it's constrained with anything there explicitly added to the build root to ensure it's consistent and reproducible.

commit 192fa42a (2020-06-08): Enable pydoc to be built without library installed

This fixes some pathing issues with the sphinx doc that previously
required the library to be installed before sphinx doc could be
correctly built. This fixes #547
PR #550 (2020-06-16): Python setup fixes
This PR fixes #547 and #545.
closed 2020-06-16 by rgetz