analogdevicesinc/libiio #548: pip install libiio error?

jaxankey, 2020-06-06:
Hello,

I cannot seem to install the python bindings with pip, which will prevent me from distributing this to students in the fall, so any help would be great!

Operating System: Linux Mint 19.3 (core is Ubuntu 18.04)
Python Distribution: Anaconda Python 3.7.3

Steps taken: 
1. I have tried installing libiio v20 and v19 via the supplied debian packages (seems to work).
2. I type `pip install libiio` and receive this error: 
```
ERROR: Could not find a version that satisfies the requirement libiio (from versions: none)
ERROR: No matching distribution found for libiio
```
I have no idea how to troubleshoot this. Sorry if this is the wrong place to post, but it seems like a packaging issue? I can't find documentation to help.

Regards,
Jack

rgetz, 2020-06-06:
Yes - sorry about that - it should have been updated yesterday for the release, but some yaml error that we didn't notice caused all the binaries/artifacts not to deploy...

Should be fixed by early next week.

jaxankey, 2020-06-06:
Ok awesome thanks for your help!!

tfcollins, 2020-06-06:
In the meantime you can grab the bindings from pypi testing:
`pip install --index-url https://test.pypi.org/simple/ libiio`

jaxankey, 2020-06-08:
Thanks, that worked!

Is there a forum for help on this? I can get ALICE to run but the scope time traces do NOT look like those in Scopy (Scopy's look like what I expect, ALICE looks like tiny wiggles + a weird offset vaguely related to the waveform I send). As such I cannot use ALICE as an example for getting reading voltage traces from the ADALM2000 in python. The AD wiki is really out of date, lots of broken links.

Thanks in advance,
Jack

tfcollins, 2020-06-08:
M2K questions go here: https://ez.analog.com/adieducation/university-program

-Travis

rgetz, 2020-06-08:
I will close this issue, I think we have things mostly resolved.

closed 2020-06-08 by rgetz