analogdevicesinc/libiio #549: Unable to create IIO context ip:

pooriayaghini, 2020-06-07:
Hi,

I am trying to work with PyAdi, but seems like it's not able to get the context through the IP.
I have tried this on both Mac and Windows. I am wondering if you can help me understand what's going wrong. Here is the error I am getting:

> PS C:\Windows\System32> .\iio_info.exe --uri 'ip:192.168.1.196'
> Library version: 0.20 (git tag: 26c5be5)
> Compiled with backends: xml ip usb serial
> Entity: line 1: parser error : Unescaped '<' not allowed in attributes values
> armv7l" ><context-attribute name="hdl_system_id" value="[fmcomms2] on [zed] git
>                                                                                ^
> Entity: line 1: parser error : attributes construct error
> armv7l" ><context-attribute name="hdl_system_id" value="[fmcomms2] on [zed] git
>                                                                                ^
> Entity: line 1: parser error : Couldn't find end of Start Tag context-attribute line 1
> armv7l" ><context-attribute name="hdl_system_id" value="[fmcomms2] on [zed] git
>                                                                                ^
> Entity: line 1: parser error : StartTag: invalid element name
> rmv7l" ><context-attribute name="hdl_system_id" value="[fmcomms2] on [zed] git <
>                                                                                ^
> ERROR: Unable to parse XML file
> Unable to create IIO context ip:192.168.1.196: Invalid argument

Thanks,
Pooria

tfcollins, 2020-06-07:
Can you provide more detail on the setup?
What version of libiio is running on the board?
What SD card version are you using on the Zed+FMComms2/3 combination?

-Travis

rgetz, 2020-06-07:
This should have been fixed with this commit:

https://github.com/analogdevicesinc/libiio/commit/71bdda598609c420c6e0bdb8e6560a517eade729

What is the library version on the host and remote?

pooriayaghini, 2020-06-08:
Here is the configuration:

On Windows 10, libiio Library version: 0.20 (git tag: 26c5be5)
On Zynq:  Ubuntu 14.04 (GNU/Linux 4.14.0-g8a57ede armv7l) + 04 February 2020 release (2019_R1 RC)

Thanks,
Pooria

rgetz, 2020-06-08:
that doesn't make sense.

If you go to the Zynq, and run iio_info there, what version is there? 

The default version of libiio on 2019_R1 will not read the /etc/libiio.ini file, which is where the hdl_system_id is.

pooriayaghini, 2020-06-09:
The bio_info seem to work fine on "local:" . But when tried to from remote terminal, or even on the Zynq with ip:127.0.0.1, it throws an error. 

Here is the iio_info output when run on Zynq:

> 
> Library version: 0.20 (git tag: 26c5be5)
> Compiled with backends: local xml ip
> IIO context created with local backend.
> Backend version: 0.20 (git tag: 26c5be5)
> Backend description string: Linux analog 4.14.0-g8a57ede #652 SMP PREEMPT Mon Feb 3 16:34:57 GMT 2020 armv7l
> IIO context has 3 attributes:
> 	hdl_system_id: [fmcomms2] on [zed] git <2e4ac278eb09c13471e381459b0da790ebad8373> clean [2019-12-04 23:47:35] UTC
> 	local,kernel: 4.14.0-g8a57ede
> 	uri: local:
> IIO context has 6 devices:
> 	iio:device0: ad7291
> 		9 channels found:
> 			voltage6:  (input)
> 			2 channel-specific attributes found:
> 				attr  0: raw value: 0
> 				attr  1: scale value: 0.610351562

rgetz, 2020-06-09:
yes - that is what described in the issue #519 - which PR #526 fixed (for me).

what I get is:

```
iio_info | less
Library version: 0.20 (git tag: 3cc5382)
Compiled with backends: local xml ip serial
IIO context created with local backend.
Backend version: 0.20 (git tag: 3cc5382)
Backend description string: Linux analog 4.19.0-g8221591d5e32 #1 SMP PREEMPT Wed May 20 14:32:17 EDT 2020 armv7l
IIO context has 9 attributes:
        hdl_system_id: [fmcomms2] on [zed] git <2e4ac278eb09c13471e381459b0da790ebad8373> clean [2019-12-04 23:47:35] UTC
        hw_model: AD-FMCOMMS2-EBZ on Xilinx Zynq ZED (armv7l)
        hw_carrier: Xilinx Zynq ZED
        hw_mezzanine: AD-FMCOMMS2-EBZ
        hw_name: AD9361 RF Hardware Development Kit
        hw_vendor: Analog Devices
        hw_serial: 00100
        local,kernel: 4.19.0-g8221591d5e32
        uri: local:
```

```
iio_info -u ip:127.0.0.1 | less
Library version: 0.20 (git tag: 3cc5382)
Compiled with backends: local xml ip serial
IIO context created with network backend.
Backend version: 0.20 (git tag: 26c5be5)
Backend description string: 127.0.0.1 Linux analog 4.19.0-g8221591d5e32 #1 SMP PREEMPT Wed May 20 14:32:17 EDT 2020 armv7l
IIO context has 10 attributes:
        hdl_system_id: [fmcomms2] on [zed] git <2e4ac278eb09c13471e381459b0da790ebad8373> clean [2019-12-04 23:47:35] UTC
        hw_model: AD-FMCOMMS2-EBZ on Xilinx Zynq ZED (armv7l)
        hw_carrier: Xilinx Zynq ZED
        hw_mezzanine: AD-FMCOMMS2-EBZ
        hw_name: AD9361 RF Hardware Development Kit
        hw_vendor: Analog Devices
        hw_serial: 00100
        local,kernel: 4.19.0-g8221591d5e32
        uri: ip:127.0.0.1
        ip,ip-addr: 127.0.0.1
```

and from a remote:

```
./tests/iio_info -u ip:192.168.1.120 | less
Compiled with backends: local xml ip usb serial
IIO context created with network backend.
Backend version: 0.20 (git tag: 26c5be5)
Backend description string: 192.168.1.120 Linux analog 4.19.0-g8221591d5e32 #1 SMP PREEMPT Wed May 20 14:32:17 EDT 2020 armv7l
IIO context has 10 attributes:
        hdl_system_id: [fmcomms2] on [zed] git <2e4ac278eb09c13471e381459b0da790ebad8373> clean [2019-12-04 23:47:35] UTC
        hw_model: AD-FMCOMMS2-EBZ on Xilinx Zynq ZED (armv7l)
        hw_carrier: Xilinx Zynq ZED
        hw_mezzanine: AD-FMCOMMS2-EBZ
        hw_name: AD9361 RF Hardware Development Kit
        hw_vendor: Analog Devices
        hw_serial: 00100
        local,kernel: 4.19.0-g8221591d5e32
        uri: ip:192.168.1.120
        ip,ip-addr: 192.168.1.120
IIO context has 8 devices:
```

Can you confirm in  context.c there is a call to `encode_xml_ndup`?

and that the version of libxml2 on your system is 2.91 or higher?

On my Zynq image: 
```
/usr/lib/arm-linux-gnueabihf/libxml2.so.2.9.1
```

on my host:
```
/usr/lib/x86_64-linux-gnu/libxml2.so.2.9.4
```

rgetz, 2020-06-09:
also to confirm - when you updated the zynq board - you rebooted things?

pooriayaghini, 2020-06-14:
Thanks Robin, Rebooting fixed the issue.

closed 2020-06-14 by pooriayaghini

rgetz, 2020-06-17:
OK - for future reference - just installing will not stop/restart the existing/running iiod, which will have the old libiio mmapped into memory. you need to restart iiod.

`make install` is not the same as `/etc/init.d/iiod restart`