analogdevicesinc/libiio #552: Tag v0.20 has been removed

dvzrv, 2020-06-09:
Hi! We are packaging `v0.20` on Arch Linux already, but it seems you have deleted the `v0.20` tag: https://github.com/analogdevicesinc/libiio/tags
This breaks downstream reproducibility and packaging.
Is there a particular reason for removing the tag?
How will we now know which commit has been used for the tag `v0.20`?

rgetz, 2020-06-09:
Yeah, sorry about that.

The code will stay the same - CI was broken (and we didn't noticed since there are portions of CI that are only run during a tag), and we are just fixing artifact releases, and some logic around the way Python Doc is built (based on feedback from someone fro the Arch team).

Hope to have things resolved very shortly.

FFY00, 2020-06-09:
@rgetz I had already fixed #545 locally (it's a bad fix so I didn't send a patch). When something like this happens it's much better to make a bugfix release, `v0.20.1`.

rgetz, 2020-06-09:
Unfortunately - the versions in libiio are set for n.n.githash; so there isn't any possibility of doing a .1 release...

We needed to fix the artifacts so people could download binaries - I know this is kind of terrible either way.

rgetz, 2020-06-17:
v0.21 is in progress of being released today, so closing this.

closed 2020-06-17 by rgetz