analogdevicesinc/libiio #555: 0.21: cmake fails with WITH_MAN=ON

kloczek, 2020-06-17:
```
+ /usr/bin/cmake -D BUILD_SHARED_LIBS=ON -D CMAKE_AR=/usr/bin/gcc-ar -D CMAKE_BUILD_TYPE=RelWithDebInfo -D CMAKE_C_FLAGS_RELEASE=-DNDEBUG -D CMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -D CMAKE_Fortran_FLAGS_RELEASE=-DNDEBUG -D CMAKE_INSTALL_PREFIX=/usr -D CMAKE_NM=/usr/bin/gcc-nm -D CMAKE_RANLIB=/usr/bin/gcc-ranlib -D CMAKE_VERBOSE_MAKEFILE=ON -D INCLUDE_INSTALL_DIR=/usr/include -D LIB_INSTALL_DIR=/usr/lib64 -D LIB_SUFFIX=64 -D SHARE_INSTALL_PREFIX=/usr/share -D SYSCONF_INSTALL_DIR=/etc . -D INSTALL_UDEV_RULE=ON -D PYTHON_BINDINGS=ON -D WITH_DOC=ON -D WITH_MAN=ON -D WITH_SERIAL_BACKEND=ON -D WITH_SYSTEMD=ON -D WITH_SYSVINIT=OFF -D WITH_TESTS=ON -D WITH_USB_BACKEND=ON -D WITH_XML_BACKEND=ON
-- cmake version: 3.17.3
-- The C compiler identification is GNU 10.1.1
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test HAS_WPEDANTIC
-- Performing Test HAS_WPEDANTIC - Success
-- Looking for strdup
-- Looking for strdup - found
-- Looking for strndup
-- Looking for strndup - found
-- Looking for strerror_r
-- Looking for strerror_r - found
-- Looking for newlocale
-- Looking for newlocale - found
-- Looking for pthread_setname_np
-- Looking for pthread_setname_np - found
-- Looking for in6addr_any
-- Looking for in6addr_any - found
-- Looking for libusb-1.0 : Found
-- Looking for libusb_get_version
-- Looking for libusb_get_version - found
-- Found Git: /usr/bin/git (found version "2.27.0")
-- Looking for libserialport : Found
-- Building with Network back end support
-- Performing Test HAS_O_TMPFILE
-- Performing Test HAS_O_TMPFILE - Success
-- Performing Test WITH_NETWORK_EVENTFD
-- Performing Test WITH_NETWORK_EVENTFD - Success
-- Performing Test HAS_PIPE2
-- Performing Test HAS_PIPE2 - Success
-- Building with Avahi, a DNS SD implementation
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Looking for lzma_auto_decoder in /usr/lib64/liblzma.so
-- Looking for lzma_auto_decoder in /usr/lib64/liblzma.so - found
-- Looking for lzma_easy_encoder in /usr/lib64/liblzma.so
-- Looking for lzma_easy_encoder in /usr/lib64/liblzma.so - found
-- Looking for lzma_lzma_preset in /usr/lib64/liblzma.so
-- Looking for lzma_lzma_preset in /usr/lib64/liblzma.so - found
-- Found LibLZMA: /usr/lib64/liblzma.so (found version "5.2.5")
-- Found ZLIB: /usr/lib64/libz.so (found version "1.2.11")
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.18") found components: doxygen dot
-- Found Dot: /usr/bin/dot
-- Check for case-sensitive file systems
-- File system is case-sensitive
-- bin= lib= inc=
-- Found Python: /usr/bin/python3.8 (found version "3.8.3") found components: Interpreter
-- new
-- Python_EXECUTABLE /usr/bin/python3.8
-- Found Python: Building bindings
-- Building with Python Doc (Sphinx)
grep: ../iio.h: No such file or directory
grep: ../iio.h: No such file or directory
grep: ../iio.h: No such file or directory
CMake Error at man/CMakeLists.txt:17 (message):
  Can not build html DOC from man without man2html

-- Configuring incomplete, errors occurred!
```

kloczek, 2020-06-17:
BTW: I seee in source tree `tests/` directory but after generate Makefile files I don't see any test/check target which executes project test suite and I don't see anything in doc about that part.
How to execute the libiio test suite? :)

rgetz, 2020-06-18:
`./tests` is mis-named, they are utilities.

The man build system (incorrectly) assumes you are building inside the source directory - I will fix that for the next release.

kloczek, 2020-06-18:
OK so there is no any test suite so far? (just asking :) )

rgetz, 2020-06-21:
Not public ones - since most need attached hardware.

We are working on it with https://github.com/martinpitt/umockdev

rgetz, 2020-07-22:
#572 should fix the build error. 

Thanks for the report.

closed 2020-07-22 by rgetz