analogdevicesinc/libiio #556: 0.21: udev rules and systemd .service files are installed in wrong place

kloczek, 2020-06-17:
cmake is executed with set of settings:
`+ /usr/bin/cmake -D BUILD_SHARED_LIBS=ON -D CMAKE_AR=/usr/bin/gcc-ar -D CMAKE_BUILD_TYPE=RelWithDebInfo -D CMAKE_C_FLAGS_RELEASE=-DNDEBUG -D CMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -D CMAKE_Fortran_FLAGS_RELEASE=-DNDEBUG -D CMAKE_INSTALL_PREFIX=/usr -D CMAKE_NM=/usr/bin/gcc-nm -D CMAKE_RANLIB=/usr/bin/gcc-ranlib -D CMAKE_VERBOSE_MAKEFILE=ON -D INCLUDE_INSTALL_DIR=/usr/include -D LIB_INSTALL_DIR=/usr/lib64 -D LIB_SUFFIX=64 -D SHARE_INSTALL_PREFIX=/usr/share -D SYSCONF_INSTALL_DIR=/etc . -D INSTALL_UDEV_RULE=ON -D PYTHON_BINDINGS=ON -D WITH_DOC=ON -D WITH_MAN=OFF -D WITH_SERIAL_BACKEND=ON -D WITH_SYSTEMD=ON -D WITH_SYSVINIT=OFF -D WITH_TESTS=ON -D WITH_USB_BACKEND=ON -D WITH_XML_BACKEND=ON
`
Install target install .rules and .service files under /lib:
```
Install the project...
/usr/bin/cmake -P cmake_install.cmake
-- Install configuration: "RelWithDebInfo"
-- Installing: /home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/lib64/pkgconfig/libiio.pc
-- Installing: /home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/lib64/libiio.so.0.21
-- Installing: /home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/lib64/libiio.so.0
-- Installing: /home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/lib64/libiio.so
-- Installing: /home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/include/iio.h
-- Installing: /home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/share/doc/libiio0-doc/v0.21
-- Installing: /home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/share/doc/libiio0-doc/v0.21/libiio

[..]

-- Installing: /home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/lib/udev/rules.d/90-libiio.rules
-- Installing: /home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/bin/iio_genxml
-- Set runtime path of "/home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/bin/iio_genxml" to ""
-- Installing: /home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/bin/iio_info
-- Set runtime path of "/home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/bin/iio_info" to ""
-- Installing: /home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/bin/iio_attr
-- Set runtime path of "/home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/bin/iio_attr" to ""
-- Installing: /home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/bin/iio_readdev
-- Set runtime path of "/home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/bin/iio_readdev" to ""
-- Installing: /home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/bin/iio_reg
-- Set runtime path of "/home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/bin/iio_reg" to ""
-- Installing: /home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/bin/iio_writedev
-- Set runtime path of "/home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/bin/iio_writedev" to ""
-- Installing: /home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/bin/iio_adi_xflow_check
-- Set runtime path of "/home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/bin/iio_adi_xflow_check" to ""
-- Installing: /home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/bin/iio_stresstest
-- Set runtime path of "/home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/bin/iio_stresstest" to ""
``

kloczek, 2020-06-17:
It would be good to obtain those udev and service path over pkgconfig

rgetz, 2020-06-18:
this is controlled via the Cmake `-DUDEV_RULES_INSTALL_DIR=/path/'`

kloczek, 2020-06-18:
Yes but instead forcing to pass that path it is possible to read that from system pkgconfig files:
```
$ pkgconf udev --variable=udevdir
/usr/lib/udev
```
The same is with install path for .service file:
```
$ pkgconf systemd --variable=systemdsystemunitdir
/usr/lib/systemd/system
```

kloczek, 2020-06-18:
In other words handling variables like UDEV_RULES_INSTALL_DIR could be simplified.

kloczek, 2020-06-18:
When `WITH_SYSTEMD=ON` is passed necessary paths could be taken from system pkgcoinfig files automatically.

rgetz, 2020-06-18:
so you can do that today with -DUDEV_RULES_INSTALL_DIR=$(pkgconf udev --variable=udevdir)

rgetz, 2020-06-18:
BTW - I know some distributions have upgraded to pkgconf, many have not.

```
pkgconf udev --variable=udevdir
bash: pkgconf: command not found
```

we would need to find out what is being used - `pkgconf` or `pkg-config` or `ppkg-config` (the perl version) - it's better to let distribution maintainers do this - since every dist is different.

kloczek, 2020-06-20:
This is not about upgrade.
pkgconfig description profides kind of inteface between components/libraries which needs to use something and tjose which are providing something. At allows as well fully standarise all that kind of interractions during build processes.

rgetz, 2020-06-21:
and not every distribution uses pkgconfig, or pkg-config - so we can't integrated it into the build system.

That is the point I was trying to make.

kloczek, 2020-06-22:
What you mean "not every distribution uses pkgconfig"?
OS distribution? or what?

rgetz, 2020-06-22:
yes - some use [pkgconf](http://pkgconf.org/), some use [pkg-config](https://people.freedesktop.org/~dbn/pkg-config-guide.html) (Debian/Ubuntu), some use [ppkgconfig](https://metacpan.org/pod/PkgConfig). Choice is great.

I'm not going to change something for one distribution that might negatively effect the others. Sorry.

closed 2020-06-24 by rgetz

kloczek, 2020-06-25:
You don't need to know what is the exact program name as long as you you are using autocoif you need to use exact aclocal macro. The same is with cmake because cmake provides module to check pkgconfig dependencies (please do not try to reinvent the wheel to handle that by yourself)..
The same is with pkgconfiig reading variables names. (which will be needed here).
In other if you are thinking that it is here something distribution specific you are right .. you are only thinking that you are right.
And in all Linux and other U*nices distributions (and even Windows) that program is used over **exactly the same name**.