analogdevicesinc/libiio #557: 0.21: python module issue on generate documetation

kloczek, 2020-06-17:
from buuild log:
```
Generating Python binding documentation with Sphinx
cd /home/tkloczko/rpmbuild/BUILD/libiio-0.21/bindings/python && /usr/bin/cmake -E env LD_LIBRARY_PATH=:/home/tkloczko/rpmbuild/BUILD/libiio-0.21 /usr/bin/sphinx-build -b html -n -c /home/tkloczko/rpmbuild/BUILD/libiio-0.21/bindings/python/doc -d /home/tkloczko/rpmbuild/BUILD/libiio-0.21/bindings/python/doctrees -w /home/tkloczko/rpmbuild/BUILD/libiio-0.21/Spx_output_python /home/tkloczko/rpmbuild/BUILD/libiio-0.21/bindings/python/doc /home/tkloczko/rpmbuild/BUILD/libiio-0.21/html/v0.21/python
Running Sphinx v2.2.2
making output directory... done
building [mo]: targets for 0 po files that are out of date
building [html]: targets for 11 source files that are out of date
updating environment: [new config] 11 added, 0 changed, 0 removed
reading sources... [  9%] buffer
reading sources... [ 18%] channel
reading sources... [ 27%] context
reading sources... [ 36%] device
reading sources... [ 45%] examples
reading sources... [ 54%] iio_attr
reading sources... [ 63%] iio_info
reading sources... [ 72%] iio_readdev
reading sources... [ 81%] iio_writedev
reading sources... [ 90%] index
reading sources... [100%] trigger

/usr/lib/python3.8/site-packages/sphinx/util/nodes.py:151: FutureWarning:
   The iterable returned by Node.traverse()
   will become an iterator instead of a list in Docutils > 0.16.
  for classifier in reversed(node.parent.traverse(nodes.classifier)):
WARNING: autodoc: failed to import class 'Buffer' from module 'iio'; the following exception was raised:
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/sphinx/ext/autodoc/importer.py", line 32, in import_module
    return importlib.import_module(modname)
  File "/usr/lib64/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/tkloczko/rpmbuild/BUILD/libiio-0.21/bindings/python/iio.py", line 236, in <module>
    _get_backends_count = _lib.iio_get_backends_count
  File "/usr/lib64/python3.8/ctypes/__init__.py", line 386, in __getattr__
    func = self.__getitem__(name)
  File "/usr/lib64/python3.8/ctypes/__init__.py", line 391, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: /usr/bin/python3: undefined symbol: iio_get_backends_count

WARNING: autodoc: failed to import class 'Channel' from module 'iio'; the following exception was raised:
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/sphinx/ext/autodoc/importer.py", line 32, in import_module
    return importlib.import_module(modname)
  File "/usr/lib64/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/tkloczko/rpmbuild/BUILD/libiio-0.21/bindings/python/iio.py", line 236, in <module>
    _get_backends_count = _lib.iio_get_backends_count
  File "/usr/lib64/python3.8/ctypes/__init__.py", line 386, in __getattr__
    func = self.__getitem__(name)
  File "/usr/lib64/python3.8/ctypes/__init__.py", line 391, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: /usr/bin/python3: undefined symbol: iio_get_backends_count

WARNING: autodoc: failed to import class 'DataFormat' from module 'iio'; the following exception was raised:
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/sphinx/ext/autodoc/importer.py", line 32, in import_module
    return importlib.import_module(modname)
  File "/usr/lib64/python3.8/importlib/__init__.py", line 127, in import_module
[... 265 lines cut ...]
checking consistency... done
preparing documents... done
writing output... [  9%] buffer
writing output... [ 18%] channel
writing output... [ 27%] context
writing output... [ 36%] device
writing output... [ 45%] examples
writing output... [ 54%] iio_attr
writing output... [ 63%] iio_info
writing output... [ 72%] iio_readdev
writing output... [ 81%] iio_writedev
writing output... [ 90%] index
writing output... [100%] trigger

generating indices...  genindex py-modindexdone
highlighting module code... [ 50%] iio
highlighting module code... [100%] iio_writedev

writing additional pages...  searchdone
copying static files... ... done
copying extra files... done
dumping search index in English (code: en)... done
dumping object inventory... done
build succeeded, 17 warnings.
````

tfcollins, 2020-06-17:
Can you provide your build commands? This is not reproducible on Fedora 32 or Centos 7.
-Travis

kloczek, 2020-06-18:
Here is full rpm package build log:
```
[tkloczko@barrel SPECS]$ rpmbuild -ba libiio.spec
Executing(%prep): /bin/sh -e /var/tmp/rpm-tmp.u5upYR
+ umask 022
+ cd /home/tkloczko/rpmbuild/BUILD
+ cd /home/tkloczko/rpmbuild/BUILD
+ rm -rf libiio-0.21
+ /usr/bin/gzip -dc /home/tkloczko/rpmbuild/SOURCES/libiio-0.21.tar.gz
+ /usr/bin/tar -xof -
+ STATUS=0
+ '[' 0 -ne 0 ']'
+ cd libiio-0.21
+ /usr/bin/chmod -Rf a+rX,u+w,g-w,o-w .
+ RPM_EC=0
++ jobs -p
+ exit 0
Executing(%build): /bin/sh -e /var/tmp/rpm-tmp.qLCGbT
+ umask 022
+ cd /home/tkloczko/rpmbuild/BUILD
+ cd libiio-0.21
+ export 'CFLAGS=-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -flto=auto -flto-partition=none'
+ CFLAGS='-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -flto=auto -flto-partition=none'
+ export 'CXXFLAGS=-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -flto=auto -flto-partition=none'
+ CXXFLAGS='-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -flto=auto -flto-partition=none'
+ export 'FFLAGS=-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -I/usr/lib64/gfortran/modules -flto=auto -flto-partition=none'
+ FFLAGS='-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -I/usr/lib64/gfortran/modules -flto=auto -flto-partition=none'
+ export 'FCFLAGS=-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -I/usr/lib64/gfortran/modules -flto=auto -flto-partition=none'
+ FCFLAGS='-O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fstack-protector-strong -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1 -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -I/usr/lib64/gfortran/modules -flto=auto -flto-partition=none'
+ export 'LDFLAGS=-Wl,-z,relro -Wl,--as-needed  -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -flto=auto -flto-partition=none -fuse-linker-plugin'
+ LDFLAGS='-Wl,-z,relro -Wl,--as-needed  -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld -flto=auto -flto-partition=none -fuse-linker-plugin'
+ export AR=/usr/bin/gcc-ar RANLIB=/usr/bin/gcc-ranlib NM=/usr/bin/gcc-nm
+ AR=/usr/bin/gcc-ar
+ RANLIB=/usr/bin/gcc-ranlib
+ NM=/usr/bin/gcc-nm
+ export CC=gcc CXX=g++
+ CC=gcc
+ CXX=g++
+ /usr/bin/cmake -D BUILD_SHARED_LIBS=ON -D CMAKE_AR=/usr/bin/gcc-ar -D CMAKE_BUILD_TYPE=RelWithDebInfo -D CMAKE_C_FLAGS_RELEASE=-DNDEBUG -D CMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -D CMAKE_Fortran_FLAGS_RELEASE=-DNDEBUG -D CMAKE_INSTALL_PREFIX=/usr -D CMAKE_NM=/usr/bin/gcc-nm -D CMAKE_RANLIB=/usr/bin/gcc-ranlib -D CMAKE_VERBOSE_MAKEFILE=ON -D INCLUDE_INSTALL_DIR=/usr/include -D LIB_INSTALL_DIR=/usr/lib64 -D LIB_SUFFIX=64 -D SHARE_INSTALL_PREFIX=/usr/share -D SYSCONF_INSTALL_DIR=/etc . -D INSTALL_UDEV_RULE=ON -D PYTHON_BINDINGS=ON -D WITH_DOC=ON -D WITH_MAN=OFF -D WITH_SERIAL_BACKEND=ON -D WITH_SYSTEMD=ON -D WITH_SYSVINIT=OFF -D WITH_TESTS=ON -D WITH_USB_BACKEND=ON -D WITH_XML_BACKEND=ON
-- cmake version: 3.17.3
-- The C compiler identification is GNU 10.1.1
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test HAS_WPEDANTIC
-- Performing Test HAS_WPEDANTIC - Success
-- Looking for strdup
-- Looking for strdup - found
-- Looking for strndup
-- Looking for strndup - found
-- Looking for strerror_r
-- Looking for strerror_r - found
-- Looking for newlocale
-- Looking for newlocale - found
-- Looking for pthread_setname_np
[... cut ...]
-- Installing: /home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/share/doc/libiio0-doc/v0.21/python/_static/fonts/fontawesome-webfont.svg
-- Installing: /home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/share/doc/libiio0-doc/v0.21/python/_static/fonts/fontawesome-webfont.ttf
-- Installing: /home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/share/doc/libiio0-doc/v0.21/python/_static/fonts/fontawesome-webfont.woff
-- Installing: /home/tkloczko/rpmbuild/BUILDROOT/libiio-0.21-2.fc33.x86_64/usr/share/doc/libiio0-doc/v0.21/python/_static/fonts/fontawesome-webfont.woff

tfcollins, 2020-06-18:
Can you provide your spec file? and what actual distribution you are using?

-Travis

kloczek, 2020-06-18:
I'm using my own distribution. which is fedora rawhide based with at the moment about 70% of all packages rebuild with LTO and -Wl,--as-needed (and solved tons of problems which still exist in fedora).
At the moment I cannot provide access to the system where you would be able to perform some actions (maybe in 2-3 months).
You have full build log. If you want me to perform some diagnostics please let me know.

tfcollins, 2020-06-18:
So the problem is that the bindings need to have access to the library, so it needs to be on path and findable through find_library in python. I have no idea where things end up since you are setting so many variables.

-Travis

rgetz, 2020-07-22:
I'm going to close - since there haven't been any updates in a month - we will try to make this easier as we fix the cross compiling / OE issues.

closed 2020-07-22 by rgetz