analogdevicesinc/libiio #558: Release tag format changed in (v)0.21

mhennerich, 2020-06-18:
All previous releases used to have a 'v' as prefix.
such as v0.19, latest release is only called 0.21.
This causes some hick ups in buildroot. These can be fixed. 
But a consistent naming schema would be desired.

dNechita, 2020-06-18:
It was a mistake of mine, sorry about that.

Would it make sense to create a git tag "v0.21" which would point to the same SHA commit as tag "0.20"? This won't fix the github release though.

rgetz, 2020-06-18:
yes - if you can do that - it would fix things on some of the build systems.

mhennerich, 2020-06-18:
For buildroot the issue was that the build artifact name changed.
Just creating a additional tag doesn't solve this. Anyways the fix for buildroot is easy.

rgetz, 2020-06-21:
So, this is Ok to be closed?

mhennerich, 2020-06-22:
Yes close

closed 2020-06-22 by mhennerich