analogdevicesinc/libiio #559: scan failure on one backend causes failures on all

rgetz, 2020-06-22:
@mhennerich noticed that an ethernet scan failure causes a total scan failure...

I think this is a problem with bad actors on ethernet with a Windows clients.

But will need to check a little bit more...

rgetz, 2020-07-22:
right now, we have `iio_scan_context_get_info_list()` which returns either:
```
 * @returns On success, the number of contexts found.
 * @returns On failure, a negative error number.
```

So, in the case we have a partial failure (usb works, but Ethernet does not), how do we want to return things?
- on partial success, just return the number of context found, and don't tell anyone Ethernet failed?

tfcollins, 2020-07-23:
What is an Ethernet failure? Is that just it didn't find anything?

commodo, 2020-07-23:
So, if it's a bad-actor on the network, we can consider it doesn't exist and continue with the rest that do exist?
We can probably identify a few error codes that make sense to ignore in the network iteration code.

Another option is to extend the API a bit, and add a `iio_scan_context_allow_partial_results_set(true)` [or some other function call that may be made more extensible, like an `iio_scan_context_ioctl()` or something], to allow for partial results.

This could probably be useful for local contexts as well [later] when stuff gets added/extended to the local context to do more validation in there.
Right now, `local_context_scan()` does something like:
```
static int check_device(void *d, const char *path)
{
        *(bool *)d = true;
        return 0;
}

int local_context_scan(struct iio_scan_result *scan_result)
{
        struct iio_context_info **info;
        bool exists = false;
        char *desc, *uri, *machine, buf[2 * BUF_SIZE], names[BUF_SIZE];
        int ret;

        ret = foreach_in_dir(&exists, "/sys/bus/iio", true, check_device);
        if (ret < 0 || !exists)
                return 0;
.........................
```

So, `check_device()` looks fine for a prototype code, but I think we can [probably] squeeze more sysfs ABI validation in there.
That could break some backwards compatibility [at first], so this should probably be kept as a backwards compatible behavior.

rgetz, 2020-07-23:
> What is an Ethernet failure? Is that just it didn't find anything?

It's when something response that they do have a IIO device via dns-sd, but don't actually have iiod running.

This happens if you have an old install with a iio_services file in /etc/avahai/ ... but no iiod running. The Linux side handles this case, but the windows implementation doesn't like it, and returns a failure.

rgetz, 2020-08-04:
I think the issue is - the Linux code actually port knocks, and the Windows code does not...

That is the easiest fix.

commit 0f234cf0 (2022-02-23): Windows DNS SD : add port knocking like other implementations

Both the Linux and Mac implemations of DNS SD ensures something is there
before passing the host information up the stack. The Windows
implemnation was missing this, so add it.

Fix #559
commit 3e1366fa (2022-02-23): Windows DNS SD : add port knocking like other implementations

Both the Linux and Mac implemations of DNS SD ensures something is there
before passing the host information up the stack. The Windows
implemnation was missing this, so add it. Add locking since it's needed
by those functions.

Fix #559
commit 34b28848 (2022-02-27): Windows DNS SD : add port knocking like other implementations

Both the Linux and Mac implemations of DNS SD ensures something is there
before passing the host information up the stack. The Windows
implemnation was missing this, so add it. Add locking since it's needed
by those functions.

Fix #559
commit a0621b64 (2022-02-27): Windows DNS SD : add port knocking like other implementations

Both the Linux and Mac implemations of DNS SD ensures something is there
before passing the host information up the stack. The Windows
implemnation was missing this, so add it. Add locking since it's needed
by those functions.

Fix #559
closed 2022-03-03 by rgetz