analogdevicesinc/libiio #560: Add UART support for IIOD

tfcollins, 2020-06-24:
Request: https://ez.analog.com/linux-software-drivers/f/q-a/167536/iio-oscilloscope-through-serial-uart-backend-on-custom-board

rgetz, 2020-07-04:
When doing this - it would be nice to return a list of available serial ports in the main library. This would ensure that end applications didn't need to link to libserial directly, and would just use libiio...

something like: `ret = iio_serialports_get_list(&info)`

and implementing that with something like:

```
struct sp_port **ports;
/*get serial ports*/
enum sp_return error = sp_list_ports(&ports);
for (i = 0; ports[i]; i++) {
      sp_get_port_name(ports[i]);
}

```

rgetz, 2020-07-04:
and it looks like - trying to open a context on a non-connected serial port hangs forever - there should be a timeout/failure mechanism of some sort...

amiraeva, 2020-08-13:
This may be out of scope for this, but I'm interested in interfacing with a client running tinyiiod using UART over BLE. At first glance, my use case might be feasible if there was a way for me to give libiio a byte sink and byte source (similar to `tinyiiod_ops` `read/write` fields). A generic byte sink/source plugin interface for iiod would also be nice if I decided to run my project on a linux device instead of a MCU.

rgetz, 2020-08-27:
tinyiiod using UART over BLE should work out of the box, and any serial client should work with that - it's already byte oriented...

(unless I'm missing something?)

amiraeva, 2020-08-27:
> tinyiiod using UART over BLE should work out of the box

Yes that was my understanding as well. What's not quite clear to me (sorry if its in the docs somewhere!) is how to interface with a client running tinyiiod over a non-standard interface such as BLE.

The only solution I can think of is to take the BLE UART code on my host, and somehow hook it up to an emulated serial port (maybe/hopefully from userspace?). Then I can get libiio to talk to the client (running tinyiiod) over BLE by using the fake serial port as a bridge.

(or am I completely missing something?)

pcercuei, 2021-04-01:
I'm pretty sure IIOD can already be used over UART without any additional code. It should just be a matter of running `iiod -i` on the serial port's TTY (in `/etc/inittab` or systemd).

commit 54a7ca8d (2021-04-15): iiod: Support running on UART

Add a -s option to serve IIO over UART. This option requires the UART
dev node and optionally the baud/parity/bits settings to be specified
as the parameter.

For instance, this will run IIOD on the /dev/ttyS2 UART, at 57600 bps,
no parity, 8 bits per datum:

iiod -s /dev/ttyS2,57600n8

Fixes #560.
commit f27dd28c (2021-04-15): iiod: Support running on UART

Add a -s option to serve IIO over UART. This option requires the UART
dev node and optionally the baud/parity/bits settings to be specified
as the parameter.

For instance, this will run IIOD on the /dev/ttyS2 UART, at 57600 bps,
no parity, 8 bits per datum:

iiod -s /dev/ttyS2,57600n8

Fixes #560.
commit e6adfc71 (2021-04-27): iiod: Support running on UART

Add a -s option to serve IIO over UART. This option requires the UART
dev node and optionally the baud/parity/bits/stop bits/flow settings
to be specified as the parameter.

For instance, this will run IIOD on the /dev/ttyS2 UART, at 57600 bps,
no parity, 8 bits per datum, 1 stop bit and RTS/CTS flow control:

iiod -s /dev/ttyS2,57600,8n1r

Fixes #560.
commit e9cda4cc (2021-04-28): iiod: Support running on UART

Add a -s option to serve IIO over UART. This option requires the UART
dev node and optionally the baud/parity/bits/stop bits/flow settings
to be specified as the parameter.

For instance, this will run IIOD on the /dev/ttyS2 UART, at 57600 bps,
no parity, 8 bits per datum, 1 stop bit and RTS/CTS flow control:

iiod -s /dev/ttyS2,57600,8n1r

Fixes #560.
closed 2021-10-01 by pcercuei