analogdevicesinc/libiio #562: Scopy doesn't find adalm2000 / "Unable to create Local IIO context : Function not implemented"

jpnielsen, 2020-07-05:
I can't get SCOPY to work on windows 10

I used the latest "green" SCOPY from here:
https://ci.appveyor.com/api/buildjobs/a1k5a7g0t9c78tly/artifacts/scopy-64-setup.exe

Refering to the intructions on [this](https://wiki.analog.com/university/tools/pluto/drivers/windows) page, i run the commands below:

Note that the wiki page doesn't instruct me to install libiio before running iio_info, but i figured that out and got it from here:
https://github.com/analogdevicesinc/libiio/releases/download/0.21/libiio-0.21.g565bf68-Windows-setup.exe

```
C:\Users\xyz>iio_info -s
Library version: 0.21 (git tag: 565bf68)
Compiled with backends: xml ip usb serial
Unable to create Local IIO context : Function not implemented
Available contexts:
        0: 0456:b672 (Analog Devices Inc. M2k (ADALM-2000)), serial=104473ce69910018f5ff070045f4884c0e [usb:2.5.5]

C:\Users\xyz>iio_info -a -C
Library version: 0.21 (git tag: 565bf68)
Compiled with backends: xml ip usb serial
Using auto-detected IIO context at URI "usb:2.5.5"
iio_info: unknown option -- C
Unknown argument '?'
```

jpnielsen, 2020-07-05:
I removed libiio 0.21 and installed libiio 0.19

```
C:\Users\xyz>iio_info -s
Library version: 0.19 (git tag: 5f5af2e)
Compiled with backends: xml ip usb serial
Available contexts:
        0: 0456:b672 (Analog Devices Inc. M2k (ADALM-2000)), serial=104473ce69910018f5ff070045f4884c0e [usb:2.2.5]

C:\Users\xyz>iio_attr -a -C
Using auto-detected IIO context at URI "usb:2.2.5"
IIO context with 20 attributes:
hw_model: Analog Devices M2k Rev.D (Z7010)
hw_model_variant: 0
hw_serial: 104473ce69910018f5ff070045f4884c0e
fw_version: v0.25
cal,offset_pos_dac: -0.0172
cal,gain_pos_dac: 1.01192
cal,offset_pos_adc: 0.00154676
cal,gain_pos_adc: 0.998392
cal,offset_neg_dac: -0.0005
cal,gain_neg_dac: 1.00518
cal,offset_neg_adc: -0.00260134
cal,gain_neg_adc: 1.00508
local,kernel: 4.14.0-gbfb0c427ad77
usb,idVendor: 0456
usb,idProduct: b672
usb,release: 2.0
usb,vendor: Analog Devices Inc.
usb,product: M2k (ADALM-2000)
usb,serial: 104473ce69910018f5ff070045f4884c0e
usb,libusb: 1.0.19.10905
```

rgetz, 2020-07-06:
> C:\Users\xyz>iio_info -a -C
> Library version: 0.21 (git tag: 565bf68)
> Compiled with backends: xml ip usb serial
> Using auto-detected IIO context at URI "usb:2.5.5"
> iio_info: unknown option -- C
> Unknown argument '?'

try `iio_attr -a -C`

jpnielsen, 2020-07-06:
oops, much better.

 iio_attr.exe -a -C
Using auto-detected IIO context at URI "usb:2.2.5"
IIO context with 20 attributes:
hw_model: Analog Devices M2k Rev.D (Z7010)
hw_model_variant: 0
hw_serial: 104473ce69910018f5ff070045f4884c0e
fw_version: v0.25
cal,offset_pos_dac: -0.0172
cal,gain_pos_dac: 1.01192
cal,offset_pos_adc: 0.00154676
cal,gain_pos_adc: 0.998392
cal,offset_neg_dac: -0.0005
cal,gain_neg_dac: 1.00518
cal,offset_neg_adc: -0.00260134
cal,gain_neg_adc: 1.00508
local,kernel: 4.14.0-gbfb0c427ad77
usb,idVendor: 0456
usb,idProduct: b672
usb,release: 2.0
usb,vendor: Analog Devices Inc.
usb,product: M2k (ADALM-2000)
usb,serial: 104473ce69910018f5ff070045f4884c0e
usb,libusb: 1.0.19.10905

However, Scopy can't see the module.
It works fine under linux on another PC running debian sid and the latest scopy flatpak.

jpnielsen, 2020-07-06:
Library version: 0.21
```
PS C:\Users\xyz> iio_info.exe -s
Library version: 0.21 (git tag: 565bf68)
Compiled with backends: xml ip usb serial
**Unable to create Local IIO context : Function not implemented**
Available contexts:
        0: 0456:b672 (Analog Devices Inc. M2k (ADALM-2000)), serial=104473ce69910018f5ff070045f4884c0e [usb:2.2.5]
PS C:\Users\xyz> iio_attr.exe -a -C
Using auto-detected IIO context at URI "usb:2.2.5"
IIO context with 21 attributes:
hw_model: Analog Devices M2k Rev.D (Z7010)
hw_model_variant: 0
hw_serial: 104473ce69910018f5ff070045f4884c0e
fw_version: v0.25
cal,offset_pos_dac: -0.0172
cal,gain_pos_dac: 1.01192
cal,offset_pos_adc: 0.00154676
cal,gain_pos_adc: 0.998392
cal,offset_neg_dac: -0.0005
cal,gain_neg_dac: 1.00518
cal,offset_neg_adc: -0.00260134
cal,gain_neg_adc: 1.00508
local,kernel: 4.14.0-gbfb0c427ad77
uri: usb:2.2.5
usb,idVendor: 0456
usb,idProduct: b672
usb,release: 2.0
usb,vendor: Analog Devices Inc.
usb,product: M2k (ADALM-2000)
usb,serial: 104473ce69910018f5ff070045f4884c0e
usb,libusb: 1.0.19.10905
```

jpnielsen, 2020-07-06:
My error.

The latest release (v1.1.2) of scopy works - I had it mixed up with the latest passing nightly build, that doesn't work.

Closing.

closed 2020-07-06 by jpnielsen