analogdevicesinc/libiio #569: cross compile for window on ubuntu  error: no TunnelType member in struct IPADAPTER_ADDRESS

anzyelay, 2020-07-16:
https://github.com/analogdevicesinc/libiio/blob/740b3f49b82be96b0a10620adcf7741eec3cd4c3/dns_sd_windows.c#L82

when i cross compile it ,i found the definitions _WIN32_WINNT means XP, so the struct  in iptypes.h pointer to an XP struct which has no  member TunnelType.

anzyelay, 2020-07-16:
i  have resolved it in  temporary as follow:

diff --git a/CMakeLists.txt b/CMakeLists.txt
index c044fe0..121ad48 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -312,6 +312,7 @@ if(WITH_NETWORK_BACKEND)
                list(APPEND LIBS_TO_LINK ${AVAHI_LIBRARIES})
        elseif(WIN32)
                set(HAVE_DNS_SD ON)
+               set(CMAKE_C_FLAGS "-D_WIN32_WINNT=0x0600 ${CMAKE_C_FLAGS}")
                list(APPEND LIBIIO_CFILES dns_sd_windows.c dns_sd.c)
                if (CMAKE_COMPILER_IS_GNUCC OR CMAKE_C_COMPILER_ID STREQUAL "Clang")
                        set_source_files_properties(dns_sd_windows.c PROPERTIES COMPILE_FLAGS "-Wno-unused-function")
diff --git a/mdns.h b/mdns.h
index 58a5bb5..7e164f7 100644
--- a/mdns.h
+++ b/mdns.h
@@ -22,8 +22,8 @@
 
 #include <fcntl.h>
 #ifdef _WIN32
-#include <Winsock2.h>
-#include <Ws2tcpip.h>
+#include <winsock2.h>
+#include <ws2tcpip.h>
 #define strncasecmp _strnicmp
 #else
 #include <unistd.h>

rgetz, 2020-07-22:
Thanks for the report.

`_WIN32_WINNT` should be set by the [compiler](https://docs.microsoft.com/en-us/cpp/porting/modifying-winver-and-win32-winnt?view=vs-2019) where `_WIN32_WINNT >= 0x600` means one of:
```
#define _WIN32_WINNT_WIN6                   0x0600 // Windows Vista
#define _WIN32_WINNT_VISTA                  0x0600 // Windows Vista
#define _WIN32_WINNT_WS08                   0x0600 // Windows Server 2008
#define _WIN32_WINNT_LONGHORN               0x0600 // Windows Vista
#define _WIN32_WINNT_WIN7                   0x0601 // Windows 7
#define _WIN32_WINNT_WIN8                   0x0602 // Windows 8
#define _WIN32_WINNT_WINBLUE                0x0603 // Windows 8.1
#define _WIN32_WINNT_WINTHRESHOLD           0x0A00 // Windows 10
#define _WIN32_WINNT_WIN10                  0x0A00 // Windows 10
```
None of those should mean:
```
#define _WIN32_WINNT_WINXP                  0x0501 // Windows XP
```

maybe your cross compiler is set up wrong? Can you describe your build system more?

We build naively with mingw and MSStudio - and have not run into this.

dNechita, 2020-07-23:
I have run into this issue when cross compiling iio-oscilloscope on Ubuntu 18.04 for Windows. What I did, was to set _WIN32_WINNT while generating the build system (without changing the CMakefiles.txt content). See here (https://github.com/analogdevicesinc/iio-osc-mingw/commit/5908ee788e413f8f19f93dd92ac0c9ffe6d5cc3d)

rgetz, 2020-07-23:
[MinGW Doc](http://www.mingw.org/wiki/Use_more_recent_defined_functions) does indicated these need to be set by hand before including windows.h header file. Shouldn't we then just include this in the CMake based on `IF(MINGW)` ? [Cmake Doc](https://cmake.org/cmake/help/v3.2/variable/MINGW.html)

dNechita, 2020-07-23:
Yes, I think we should include these definitions when MINGW is used since the dns_sd code for windows won't compile if _WIN32_WINNT points to some old versions of Windows.

rgetz, 2020-07-23:
I think this is now fixed in #574

commit f17bd3ba (2020-07-23): cmake: set _WIN32_WINNT when compiling with mingw

This should fix #569
commit b80ae015 (2020-07-23): cmake: set _WIN32_WINNT when compiling with mingw

This should fix #569
commit d3be2ee6 (2020-07-23): cmake: set _WIN32_WINNT when compiling with mingw

This should fix #569
PR #574 (2020-07-24): cmake: set _WIN32_WINNT when compiling with mingw
This should fix #569
closed 2020-07-23 by rgetz