analogdevicesinc/libiio #573: iio_attr cannot take negative numbers

tfcollins, 2020-07-22:
```
root@analog:~# iio_attr -o -c adrv9002-phy voltage0 hardwaregain
-20.000000 dB
root@analog:~# iio_attr -o -c adrv9002-phy voltage0 hardwaregain -10
iio_attr: invalid option -- '1'
Unknown argument '?'
root@analog:~# iio_attr -o -c adrv9002-phy voltage0 hardwaregain "-10"
iio_attr: invalid option -- '1'
Unknown argument '?'
root@analog:~# iio_attr -o -c adrv9002-phy voltage0 hardwaregain 0
0.000000 dB
root@analog:~# 

```
It does not know when to stop looking for flags.

Workaround:

```
root@analog:~# iio_attr -o -c -- adrv9002-phy voltage0 hardwaregain -10
-10.000000 dB
```

rgetz, 2020-07-28:
confirmed - and I have a patch I'm testing locally...

rgetz, 2020-07-28:
I made a PR #577 - so closing this for now.

closed 2020-07-28 by rgetz