analogdevicesinc/libiio #579: Evaluate & reduce libiio's stack-usage

commodo, 2020-07-30:
musl-libc [which has gained some popularity in recent years] used to allocate 16k for thread-stack, then 80k, now 128k;
http://git.musl-libc.org/cgit/musl/commit/?id=13b3645c46518e0e2fb91f0d665e60f7a8a44b2b
http://git.musl-libc.org/cgit/musl/commit/?id=c0058ab465e950c2c3302d2b62e21cc0b494224b

glibc's default is 1M [or 2M?]; anyway, it's way bigger;

depending on which version of musl is used, this can be a problem; it's still a bit likely that some deployment use 80k;

issue is with bits of code that do:
```
char err_str[1024];
iio_strerror(errno, err_str, sizeof(err_str));
```

These `err_str` may be too large. Maybe 32 is enough?
Depending on how the user structures the code, libiio can be responible for a stack-overflow error.
Which can be the user's fault, but even so, libiio should not be such a big user of stack-memory.

If 32 bytes [or even 16/24, doesn't need to be a round number] is enough to store these error strings, then I'd say to do reduce to 32 bytes [or less], and that's perfect.
If we need to go above 64, we can do that short-term.
But we may need to consider [at some point] some smarter memory-pool management for these error strings.

rgetz, 2020-07-30:
This is a dup of #72 - isn't it?

Most of these will go away during the error reporting re-do.

-Robin

commodo, 2020-07-30:
Right, didn't see it.
Sorry for the noise

closed 2020-07-30 by commodo