analogdevicesinc/libiio #582: [Query] Libiio windows support

baburb0202, 2020-08-13:
Hi,

I would like to know libiio weather support on windows or not.
Would like to run iiod on windows and share the platform device data to device in remote (IP).

As per the architecture, i understand local backend reads data from device sysfs entries and 
transfer data to remote device.

Could please share your thoughts on this.

Thanks,
Babu

rgetz, 2020-08-14:
libiio (the client) supports Windows. That's how we get libiio applications on Windows (like MATLAB, or GNU Radio).

nominally - iiod supports local backends only, making local devices available over USB or ethernet (and soon serial). Since local backends are only supported on Linux - that means the normal use case is run iiod on a Linux machine to allow local iio devices accessible to others - anywhere they might be.

Does that answer your question?

rgetz, 2020-08-14:
There is a windows installer in the releases (if that is what you are looking for)

rgetz, 2020-08-23:
I will close this - since there hasn't been a follow up in a week. If you need anything else - please re-open.

closed 2020-08-23 by rgetz