analogdevicesinc/libiio #588: Undefined reference to `encode_xml_ndup' when XML backend is not selected

mhei, 2020-09-15:
When configuring to not use XML backend, there still seems to be a dependency to libxml2:

```
mhei@kerker:/srv/smb/libiio.git$ mkdir build
mhei@kerker:/srv/smb/libiio.git$ cd build
mhei@kerker:/srv/smb/libiio.git/build$ cmake ../ -DWITH_TESTS=OFF -DWITH_XML_BACKEND=OFF -DWITH_USB_BACKEND=OFF -DWITH_SERIAL_BACKEND=OFF -DWITH_NETWORK_BACKEND=OFF -DWITH_LOCAL_BACKEND=ON
-- cmake version: 3.10.2
-- The C compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test HAS_WPEDANTIC
-- Performing Test HAS_WPEDANTIC - Success
-- Looking for strdup
-- Looking for strdup - found
-- Looking for strndup
-- Looking for strndup - found
-- Looking for strerror_r
-- Looking for strerror_r - found
-- Looking for newlocale
-- Looking for newlocale - found
-- Looking for pthread_setname_np
-- Looking for pthread_setname_np - found
-- Looking for in6addr_any
-- Looking for in6addr_any - found
-- Looking for libusb-1.0 : Found
-- Found Git: /usr/bin/git (found version "2.17.1") 
-- Looking for libserialport : Failed; building without serial
-- Building without network support
-- Found LibXml2: /usr/lib/x86_64-linux-gnu/libxml2.so (found version "2.9.4") 
-- Found BISON: /usr/bin/bison (found version "3.0.4") 
-- Found FLEX: /usr/bin/flex (found version "2.6.4") 
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of struct usb_functionfs_descs_head_v2
-- Check size of struct usb_functionfs_descs_head_v2 - done
-- Configuring done
-- Generating done
-- Build files have been written to: /srv/smb/libiio.git/build
mhei@kerker:/srv/smb/libiio.git/build$ make
Scanning dependencies of target iio
[  5%] Building C object CMakeFiles/iio.dir/backend.c.o
[ 10%] Building C object CMakeFiles/iio.dir/channel.c.o
[ 15%] Building C object CMakeFiles/iio.dir/device.c.o
[ 20%] Building C object CMakeFiles/iio.dir/context.c.o
[ 25%] Building C object CMakeFiles/iio.dir/buffer.c.o
[ 30%] Building C object CMakeFiles/iio.dir/utilities.c.o
[ 35%] Building C object CMakeFiles/iio.dir/scan.c.o
[ 40%] Building C object CMakeFiles/iio.dir/sort.c.o
[ 45%] Building C object CMakeFiles/iio.dir/local.c.o
[ 50%] Building C object CMakeFiles/iio.dir/libini/libini.c.o
[ 55%] Building C object CMakeFiles/iio.dir/lock.c.o
[ 60%] Linking C shared library libiio.so
[ 60%] Built target iio
[ 65%] [FLEX][lexer] Building scanner with flex 2.6.4
[ 70%] [BISON][parser] Building parser with bison 3.0.4
Scanning dependencies of target iiod
[ 75%] Building C object iiod/CMakeFiles/iiod.dir/iiod.c.o
[ 80%] Building C object iiod/CMakeFiles/iiod.dir/ops.c.o
[ 85%] Building C object iiod/CMakeFiles/iiod.dir/thread-pool.c.o
[ 90%] Building C object iiod/CMakeFiles/iiod.dir/parser.c.o
[ 95%] Building C object iiod/CMakeFiles/iiod.dir/lexer.c.o
[100%] Linking C executable iiod
../libiio.so.0.21: undefined reference to `encode_xml_ndup'
collect2: error: ld returned 1 exit status
iiod/CMakeFiles/iiod.dir/build.make:212: recipe for target 'iiod/iiod' failed
make[2]: *** [iiod/iiod] Error 1
CMakeFiles/Makefile2:158: recipe for target 'iiod/CMakeFiles/iiod.dir/all' failed
make[1]: *** [iiod/CMakeFiles/iiod.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

rgetz, 2020-09-17:
I will have a look - but since all the internals of libiio are built around xml  - I'm not sure what functionality will be there when you don't have xml supported...

mhei, 2020-09-18:
Hi Robin, I agree - but then the simplest solution would be to not allow the user to disable XML support at all, right? In other words, dropping the cmake option and the compile magic and always build and link this material...

hellow554, 2021-01-25:
This is still an issue. Any updates on this @rgetz ?

PR #659 (2021-04-07): xml: Remove unused function encode_xml_ndup()
This function is not used anywhere anymore since commit 9d79757.

Fixes #588.
commit fdb2ae9a (2021-04-07): xml: Remove unused function encode_xml_ndup()

This function is not used anywhere anymore since commit 9d79757.

Fixes #588.
closed 2021-04-07 by pcercuei