analogdevicesinc/libiio #591: Reproducible Builds

rgetz, 2020-09-17:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=970498

tfcollins, 2020-09-17:
Anyway for us to test this?

rgetz, 2020-09-18:
make two builds - make sure they are identical ?

mhennerich, 2020-09-21:
@dNechita  can you apply the patch from here:
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=970498

and test reproducibility?

dNechita, 2020-09-21:
Yes, I will test.

dNechita, 2020-09-22:
I've tested with the changes in the patch
 What the patch does is that allows setting the date during the cmake step. So, if you want to reproduce an identical build with a yesterday build, all you need is to set an environment variable with the yesterday date and then do the build.
Then, the man pages will be identical.

dNechita, 2020-09-24:
The idea is to add the changes in the patch to libiio, right?
If so, I'm not sure how to apply the patch (which seems to be a debian diff sort of thing) in order the preserve the original author. Any suggestions?

rgetz, 2020-09-28:
I think it is just : 
`git commit --author="Author Name <email@address.com>"`

but you need their permission first. The DCO is per author, and we won't accept anything without a signed-off-by by the original author.

-Robin

lamby, 2020-10-10:
Added PR. :+1:

closed 2020-10-11 by commodo

rgetz, 2020-10-11:
Thanks for the PR

commit 53ad0a5e (2020-10-11): Make the build reproducible (Closes: #591)

Whilst working on the Reproducible Builds effort [0] we noticed that
libiio could not be built reproducibly.

This is because it uses the current build time in the generated
manpages. Patch attached that uses SOURCE_DATE_EPOCH [1].

 [0] https://reproducible-builds.org/
 [1] https://reproducible-builds.org/docs/source-date-epoch/