analogdevicesinc/libiio #592: Cross Compilation (through Yocto) Fails With Python Bindings Enabled

mnsgs, 2020-09-28:
Since the release of https://github.com/analogdevicesinc/libiio/releases/tag/0.21 (https://github.com/analogdevicesinc/libiio/releases/tag/v0.21-rc.1 unknown, no issues on https://github.com/analogdevicesinc/libiio/releases/tag/v0.19) compilation fails when Python bindings enabled. Initial (but stale) discussion around the recipe was brought up here: https://github.com/openembedded/meta-openembedded/issues/248, however not yet resolved. I am bringing it up here, as it is not clear to me whether the resolution is within the library or the recipe. AFAICS, two issues are present:

 * setuptools looks for setup.py even before the native compilation (I assume file is rendered using cmake) 
 * setuptools looks for setup.py in the root of the source folder (resides under build/bindings/python/ for a native comnpilation)

I guess the root cause is related to the PYPI packaging of the v0.21 release. Not really sure how to proceed, however as a beginning I hope ADI can reproduce and potentially ack the issue? AFAICS from meta-adi (kudos for merging https://github.com/analogdevicesinc/meta-adi/commit/d973914dd551c9f891824665e5793154939a547d btw.) this layer does not enable the bindings, but perhaps it ought to in order to avoid regressions like this?

I am working out from a pure Yocto (non-petalinux) setup, with the local changes:

```
diff --git a/meta-oe/recipes-support/libiio/libiio_git.bb b/meta-oe/recipes-support/libiio/libiio_git.bb
index b308bde17..a619a4ed3 100644
--- a/meta-oe/recipes-support/libiio/libiio_git.bb
+++ b/meta-oe/recipes-support/libiio/libiio_git.bb
@@ -4,7 +4,7 @@ SECTION = "libs"
 LICENSE = "LGPLv2.1+"
 LIC_FILES_CHKSUM = "file://COPYING.txt;md5=7c13b3376cea0ce68d2d2da0a1b3a72c"
 
-SRCREV = "565bf68eccfdbbf22cf5cb6d792e23de564665c7"
+SRCREV = "68ab31448e108cc13de1a45744fd487021c48f15"
 PV = "0.21+git${SRCPV}"
 
 SRC_URI = "git://github.com/analogdevicesinc/libiio.git;protocol=https"
@@ -25,13 +25,14 @@ EXTRA_OECMAKE = " \
     ${@bb.utils.contains('DISTRO_FEATURES', 'systemd', '-DWITH_SYSTEMD=ON -DSYSTEMD_UNIT_INSTALL_DIR=${systemd_system_unitdir}', '', d)} \
 "
 
-PACKAGECONFIG ??= "usb_backend network_backend"
+# Hacked way of enabling bindings just for test purpose
+PACKAGECONFIG ??= "usb_backend network_backend libiio-python3"
 
 PACKAGECONFIG[usb_backend] = "-DWITH_USB_BACKEND=ON,-DWITH_USB_BACKEND=OFF,libusb1,libxml2"
 PACKAGECONFIG[network_backend] = "-DWITH_NETWORK_BACKEND=ON,-DWITH_NETWORK_BACKEND=OFF,libxml2"
 PACKAGECONFIG[libiio-python3] = "-DPYTHON_BINDINGS=ON,-DPYTHON_BINDINGS=OFF"
 
-inherit ${@bb.utils.contains('PACKAGECONFIG', 'libiio-python3', 'distutils3-base', '', d)}
+inherit ${@bb.utils.contains('PACKAGECONFIG', 'libiio-python3', 'setuptools3', '', d)}
```

and encounter the error described at https://github.com/openembedded/meta-openembedded/issues/248.

Thanks,
Martin

julienmalik, 2020-09-28:
Hi @mnsgs 

I'm on thud, basing myself from a 0.19 recipe version, and I had to bbappend this :

```
FILESEXTRAPATHS_prepend := "${THISDIR}/files:"

SRCREV = "v0.21"
PV = "0.21+git${SRCPV}"

PACKAGECONFIG ?= "network_backend libiio-python3"

# From commit 21023a538b125dee966bf496dcad019ee20967bd, libiio depends on
# setuptools3 instead of distutils, which makes the recipe for libiio 0.19
# not fully compatible with 0.21
#
# The following comes from setuptools3.bbclass
# distutils3-base is already inherited in base libiio recipe
# Inheriting from setuptools3 here does not work,
# but setuptools3 directly inherit distutils3-base and adds this DEPENDS.
# So we just replicate the behavior
DEPENDS += "python3-setuptools-native"

SRC_URI_append += "file://0001-python-Do-not-verify-whether-libiio-is-installed-whe.patch"
```

The added patch is the content of https://github.com/analogdevicesinc/libiio/pull/576

But probably the `python3-setuptools-native` is not needed in your case. IIRC I needed it because I could not inherit from setuptools3 because the original recipe inherits distutils3-base...

For me it has been sufficient to successfully build the bindings

Hope it helps

mnsgs, 2020-09-28:
Thanks for your feedback @julienmalik. It appears that

```
-PACKAGECONFIG ??= "usb_backend network_backend"
+# Hacked way of enabling bindings just for test purpose
+PACKAGECONFIG ??= "usb_backend network_backend libiio-python3"
```

breaks the compilation, however:

```
diff --git a/meta-oe/recipes-support/libiio/libiio_git.bb b/meta-oe/recipes-support/libiio/libiio_git.bb
index b308bde17..7b3c93268 100644
--- a/meta-oe/recipes-support/libiio/libiio_git.bb
+++ b/meta-oe/recipes-support/libiio/libiio_git.bb
@@ -17,6 +17,7 @@ inherit cmake python3native systemd
 DEPENDS = " \
     flex-native bison-native libaio \
     ${@bb.utils.contains('DISTRO_FEATURES', 'zeroconf', 'avahi', '', d)} \
+    ${@bb.utils.contains('PACKAGECONFIG', 'libiio-python3', 'python3-setuptools-native', '', d)} \
 "
 
 EXTRA_OECMAKE = " \
@@ -31,7 +32,7 @@ PACKAGECONFIG[usb_backend] = "-DWITH_USB_BACKEND=ON,-DWITH_USB_BACKEND=OFF,libus
 PACKAGECONFIG[network_backend] = "-DWITH_NETWORK_BACKEND=ON,-DWITH_NETWORK_BACKEND=OFF,libxml2"
 PACKAGECONFIG[libiio-python3] = "-DPYTHON_BINDINGS=ON,-DPYTHON_BINDINGS=OFF"
 
-inherit ${@bb.utils.contains('PACKAGECONFIG', 'libiio-python3', 'distutils3-base', '', d)}
+inherit ${@bb.utils.contains('PACKAGECONFIG', 'libiio-python3', 'setuptools3', '', d)}
 
 PACKAGES =+ "${PN}-iiod ${PN}-tests ${PN}-${PYTHON_PN}"
 
@@ -46,3 +47,5 @@ FILES_${PN}-${PYTHON_PN} = "${PYTHON_SITEPACKAGES_DIR}"
 
 SYSTEMD_PACKAGES = "${PN}-iiod"
 SYSTEMD_SERVICE_${PN}-iiod = "iiod.service"
+
+PACKAGECONFIG ?= "network_backend libiio-python3"

```

enables compilation. Leaving out the `python3-setuptools-native` dependency causes the compilation error:

`ModuleNotFoundError: No module named 'setuptools'`

hence I assume the dependency addition is indeed valid. I suggest to close the issue here (since it is not related to the library) and continue the discussion at https://github.com/openembedded/meta-openembedded/issues/248 - agree?

Thanks

rgetz, 2020-09-28:
are you testing the release, or master? I think there are fixes in master for this already.

mnsgs, 2020-09-28:
@rgetz are you referring to libiio master or meta-openembedded master? I am basing libiio on https://github.com/analogdevicesinc/libiio/commit/68ab31448e108cc13de1a45744fd487021c48f15 which is a prerequisite but not sufficient to build the bindings on v0.19/v0.21-rc.1+. 

AFAICS, the recipe needs to adapt towards the introduced pypi integration (perhaps https://github.com/analogdevicesinc/libiio/commit/21023a538b125dee966bf496dcad019ee20967bd ?)

Thanks,
Martin

mnsgs, 2020-09-30:
Closing the issue and continuing the recipe chat at https://github.com/openembedded/meta-openembedded/issues/248

closed 2020-09-30 by mnsgs