analogdevicesinc/libiio #594: Stack overflow when using C# bindings

LehMaxence, 2020-10-05:
Hi all,

I get, what appears, a stack overflow after reading a device attribute multiple times:

```
        static void Main(string[] args)
        {
            try
            {
                var ctx = new Context("ip:172.16.0.2");

                for (int i = 0; i < 1126; i++)
                {
                    ctx.devices.FirstOrDefault()?.attrs.FirstOrDefault()?.read();
                }

                ctx.Dispose();
            }
            finally
            {
                Console.ReadKey();
            }
        }
```

On my machine, 1126 is the magic number when the overflow crashes the app. It might be different on yours.

I am using the image for the AD4003 on a ZedBoard connected through ethernet to my machine.

rgetz, 2020-10-07:
thanks for reporting, and for putting together the reduced test case - we will have a look.

I quickly tried:
```
for (( n=1; n<=2000; n++ )) ; do echo $n ; iio_attr -u ip:192.168.1.116 -d ad9361-phy xo_correction; done
```

that works (since it's opens/closes the context every time).

I then ran `iio_attr -g ./foo.c -u ip:192.168.1.116 -d ad9361-phy xo_correction` which generates c code for `foo.c`, added a loop to things...

```
        /* Create IIO Context */
        IIO_ASSERT(ctx = iio_create_context_from_uri("ip:192.168.1.116"));

        /* Find IIO device in current context */
        IIO_ASSERT(dev = iio_context_find_device(ctx, "ad9361-phy"));

        /* Read IIO device attribute, and put result in string */
        for (int n = 0; n < 2000; n++) {
                RET_ASSERT(ret = iio_device_attr_read(
                        dev, "xo_correction", buf, sizeof(buf)));
                printf("xo_correction : %i : %s\n", n, buf);
        }
```

which also worked - which means it's likely a C# specific issue.

we will have a look, but it might take a bit

LehMaxence, 2020-10-19:
Fixed by PR #606

closed 2020-10-19 by LehMaxence