analogdevicesinc/libiio #611: Eliminate variable shadowing in local.c

msporcich, 2020-10-19:
In local.c, there are two instances of variable shadowing which cause the build to fail when compiling with flag 
`-Werror=shadow 
`

In local_close()
`  error: declaration of ‘i’ shadows a previous local [-Werror=shadow]
`
In get_short_attr_name()
`error: declaration of ‘len’ shadows a previous local [-Werror=shadow]
`

msporcich, 2020-10-20:
Forked with a proposed resolution.  I'm interested in getting any comments and then signoff so that I may issue a pull request.  Thanks.

commodo, 2020-10-20:
the change looks reasonable to me
feel free do a PR targeting our master;

thanks
Alex

rgetz, 2020-11-07:
Only question is should we add `-Werror=shadow` all the time? (for gcc)

Nominally it's `-Wall -Wextra -Wno-unused-parameter -Wpedantic -std=c99`

rgetz, 2020-11-10:
closing this.

closed 2020-11-10 by rgetz