analogdevicesinc/libiio #616: Static build with Visual Studio

tanshihaj, 2020-10-24:
I'm wonder is it possible to build static version of libiio using Visual Studio compiler? Currently when I try to do it I get LNK2019 unresolved external symbol __imp__ errors since here https://github.com/analogdevicesinc/libiio/blob/99ef2015f4163e904cff28c3ff461a348cc9f488/iio.h#L75-L79 it defines `__api` either as `__declspec(dllexport)` or `__declspec(dllimport)`, so linker, that links this library to my test program, thinks that this symbols should be defined in `.dll`.

Usually if you need to link static library `__api` should be defined as empty:
https://github.com/substack/libssh/blob/c073979235eb0d0587ac9cb3c192e91e32d34b06/include/libssh/libssh.h#L26-L42

Is it possible to add some kind of `LIBIIO_STATIC` definition to make it possible to link libiio as static library?

tanshihaj, 2020-10-24:
I've tried to do it and it work in my case: https://github.com/tanshihaj/libiio/commit/3b9305e2d60000ee011ec133edaac46f2e133a4d

dNechita, 2020-10-27:
I think we can add an option to build libIIO as a static library.

rgetz, 2020-11-07:
Just as a reminder - libiio is LGPL. From the [GNU.org website](https://www.gnu.org/licenses/gpl-faq.en.html#LGPLStaticVsDynamic) emphasis in mine:

(1) If you statically link against an LGPLed library, **_you must also provide your application in an object_** (not necessarily source) format, **_so that a user has the opportunity to modify the library and relink the application_**.
(2) If you dynamically link against an LGPLed library already present on the user's computer, you need not convey the library's source. On the other hand, **_if you yourself convey the executable LGPLed library along with your application,_** whether linked with statically or dynamically, **_you must also convey the library's sources_**, in one of the ways for which the LGPL provides.

It's not hard/difficult - but you must comply with the obligations of the license.

PR #658 (2021-04-07): CMake: Allow building as static lib in MSVC
Set the LIBIIO_STATIC define when compiling a static library, which then
causes the __api to be resolved as empty.

This fixes building statically under MSVC.

Fixes #616.
commit 365108d8 (2021-04-07): CMake: Allow building as static lib in MSVC

Set the LIBIIO_STATIC define when compiling a static library, which then
causes the __api to be resolved as empty.

This fixes building statically under MSVC.

Fixes #616.
closed 2021-04-07 by pcercuei