analogdevicesinc/libiio #628: Problem with Python bindings for the 'clear' channel of a tcs3472 sensor

boukeas, 2020-12-11:
Hello. I am working with a TCS3472 colour sensor on a Raspberry Pi running Buster.

Having installed `python3-libiio`, any attempt to access properties of the 'clear' channel results in a segmentation fault. This is a minimal example that should reproduce the issue.
```
import iio
device = iio.Context().find_device('tcs3472')
device.find_channel('intensity_clear').enabled
```
If you substitute the 'clear' channel in the last line for any of the RGB channels, there is no error:

mhennerich, 2020-12-11:
can you run iio_info command on the target?

boukeas, 2020-12-11:
I had to install `libiio-utils` in order to do so. Here is the output:
```
Library version: 0.16 (git tag: v0.16)
Compiled with backends: local xml ip usb serial
IIO context created with local backend.
Backend version: 0.16 (git tag: v0.16)
Backend description string: Linux raspberrypi 5.4.80-v7l+ #5 SMP Tue Dec 1 20:01:54 GMT 2020 armv7l
IIO context has 1 attributes:
	local,kernel: 5.4.80-v7l+
IIO context has 1 devices:
	iio:device0: tcs3472 (buffer capable)
		5 channels found:
			intensity_clear:  (input, index: 0, format: le:U16/16>>0)
			3 channel-specific attributes found:
				attr  0: calibscale value: 1
				attr  1: integration_time value: 0.614400
				attr  2: raw value: 5804
			intensity_red:  (input, index: 1, format: le:U16/16>>0)
			3 channel-specific attributes found:
				attr  0: calibscale value: 1
				attr  1: integration_time value: 0.614400
				attr  2: raw value: 1964
			intensity_green:  (input, index: 2, format: le:U16/16>>0)
			3 channel-specific attributes found:
				attr  0: calibscale value: 1
				attr  1: integration_time value: 0.614400
				attr  2: raw value: 2308
			intensity_blue:  (input, index: 3, format: le:U16/16>>0)
			3 channel-specific attributes found:
				attr  0: calibscale value: 1
				attr  1: integration_time value: 0.614400
				attr  2: raw value: 2606
			timestamp:  (input, index: 4, format: le:S64/64>>0)
		3 device-specific attributes found:
				attr  0: calibscale_available value: 1 4 16 60
				attr  1: current_timestamp_clock value: realtime

				attr  2: integration_time_available value: 0.002400 0.004800 0.007200 0.009600 0.012000 0.014400 0.016800 0.019200 0.021600 0.024000 0.026400 0.028800 0.031200 0.033600 0.036000 0.038400 0.040800 0.043200 0.045600 0.048000 0.050400 0.052800 0.055200 0.057600 0.060000 0.062400 0.064800 0.067200 0.069600 0.072000 0.074400 0.076800 0.079200 0.081600 0.084000 0.086400 0.088800 0.091200 0.093600 0.096000 0.098400 0.100800 0.103200 0.105600 0.108000 0.110400 0.112800 0.115200 0.117600 0.120000 0.122400 0.124800 0.127200 0.129600 0.132000 0.134400 0.136800 0.139200 0.141600 0.144000 0.146400 0.148800 0.151200 0.153600 0.156000 0.158400 0.160800 0.163200 0.165600 0.168000 0.170400 0.172800 0.175200 0.177600 0.180000 0.182400 0.184800 0.187200 0.189600 0.192000 0.194400 0.196800 0.199200 0.201600 0.204000 0.206400 0.208800 0.211200 0.213600 0.216000 0.218400 0.220800 0.223200 0.225600 0.228000 0.230400 0.232800 0.235200 0.237600 0.240000 0.242400 0.244800 0.247200 0.249600 0.252000 0.254400 0.256800 0.259200 0.261600 0.264000 0.266400 0.268800 0.271200 0.2736
		2 buffer-specific attributes found:
				attr  0: data_available value: 0
				attr  1: watermark value: 1
```

boukeas, 2020-12-11:
Here is some debugger info to contrast the "blue" and "clear" channels:
```
blue = {dev = 0x4abff8, pdata = 0x447008, userdata = 0x0, is_output = false, is_scan_element = true, format = {length = 16, bits = 16, shift = 0, is_signed = false, is_fully_defined = true, is_be = false, with_scale = false, scale = 0, repeat = 1}, name = 0x0, id = 0x4b02f0 "", index = 3, modifier = IIO_MOD_LIGHT_BLUE, type = IIO_INTENSITY, attrs = 0x459288, nb_attrs = 3, number = 3}

blue->dev = {ctx = 0x4b6e60, pdata = 0x447008, userdata = 0x0, name = 0x100 <error: Cannot access memory at address 0x100>, id = 0x10 <error: Cannot access memory at address 0x10>, attrs = 0x10, nb_attrs = 0, buffer_attrs = 0x100,
nb_buffer_attrs = 0, debug_attrs = 0x0, nb_debug_attrs = 1, channels = 0x0, nb_channels = 0, mask = 0x46d138, words = 2}
```
```
clear = {dev = 0x459378, pdata = 0x447008, userdata = 0x0, is_output = false, is_scan_element = true, format = {length = 16, bits = 16, shift = 0, is_signed = false, is_fully_defined = true, is_be = false, with_scale = false, scale = 0,
repeat = 1}, name = 0x0, id = 0x451c28 "h\353F", index = 0, modifier = IIO_MOD_LIGHT_CLEAR, type = IIO_INTENSITY, attrs = 0x4592a8, nb_attrs = 3, number = 0}

clear->dev = {ctx = 0x0, pdata = 0x447008, userdata = 0x6e206574, name = 0x3d656d61 <error: Cannot access memory at address 0x3d656d61>, id = 0x746e6922 <error: Cannot access memory at address 0x746e6922>, attrs =
0x61726765, nb_attrs = 1852795252, buffer_attrs = 0x6d69745f, nb_buffer_attrs = 1713381989, debug_attrs = 0x6e656c69, nb_debug_attrs = 1030057313, channels = 0x5f6e6922, nb_channels = 1702129257, mask = 0x7469736e, words = 1852399481}
```
The `ctx` context in the latter case seems to be set to null?

rgetz, 2020-12-13:
Thanks for reporting/looking into things.

v0.16 is pretty old (21 Nov 2018). Alot has changed/been updated since then.

The first thing to test would be to see if it's a fixed issue. Can you try something more recent? ( releases are: [Raspbian-8](https://github.com/analogdevicesinc/libiio/releases/download/v0.21/libiio-0.21.g565bf68-raspbian-8-armhf.deb) and [Raspbian-9](libiio-0.21.g565bf68-raspbian-9-armhf.deb) ) There were alot of changes/updates/fixes in the python bindings in 0.20

next step would be try master (just in case there were some recent issues fixed).

next step is to do a similar thing with `iio_attr`, to see if it is a problem in the library, or the driver.

If you then think its a problem in the library - let us know. If you think it's a problem in the driver - then you can check out the iio kernel mailing list.

Thanks
-Robin

XECDesign, 2020-12-14:
I have the same problem replicated on my pi and am still seeing it with the latest build from master.

```
$ iio_attr  -c tcs3472
DEBUG: Added attr 'in_intensity_calibscale' to device 'iio:device0'
DEBUG: Added attr 'raw' to channel 'intensity_green'
DEBUG: Added input channel 'intensity_green' to device 'iio:device0'
DEBUG: Added attr 'calibscale_available' to device 'iio:device0'
DEBUG: Added attr 'integration_time_available' to device 'iio:device0'
DEBUG: Added attr 'in_intensity_integration_time' to device 'iio:device0'
DEBUG: Added attr 'raw' to channel 'intensity_blue'
DEBUG: Added input channel 'intensity_blue' to device 'iio:device0'
DEBUG: Added attr 'current_timestamp_clock' to device 'iio:device0'
DEBUG: Added attr 'raw' to channel 'intensity_clear'
DEBUG: Added input channel 'intensity_clear' to device 'iio:device0'
DEBUG: Added attr 'raw' to channel 'intensity_red'
DEBUG: Added input channel 'intensity_red' to device 'iio:device0'
DEBUG: Added buffer attr 'watermark' to device 'iio:device0'
DEBUG: Added buffer attr 'data_available' to device 'iio:device0'
DEBUG: Add protected attr 'en' to channel 'intensity_blue'
DEBUG: Add protected attr 'en' to channel 'intensity_clear'
DEBUG: Add protected attr 'type' to channel 'intensity_blue'
DEBUG: Add protected attr 'en' to channel 'intensity_green'
DEBUG: Add protected attr 'index' to channel 'intensity_clear'
DEBUG: Add protected attr 'index' to channel 'timestamp'
DEBUG: Added input channel 'timestamp' to device 'iio:device0'
DEBUG: Add protected attr 'index' to channel 'intensity_red'
DEBUG: Add protected attr 'en' to channel 'intensity_red'
DEBUG: Add protected attr 'en' to channel 'timestamp'
DEBUG: Add protected attr 'index' to channel 'intensity_green'
DEBUG: Add protected attr 'type' to channel 'intensity_red'
DEBUG: Add protected attr 'type' to channel 'timestamp'
DEBUG: Add protected attr 'type' to channel 'intensity_green'
DEBUG: Add protected attr 'type' to channel 'intensity_clear'
DEBUG: Add protected attr 'index' to channel 'intensity_blue'
DEBUG: Found match: intensity_green and intensity_calibscale
DEBUG: Found match: intensity_blue and intensity_calibscale
DEBUG: Found match: intensity_clear and intensity_calibscale
DEBUG: Found match: intensity_red and intensity_calibscale
DEBUG: Found match: intensity_green and intensity_calibscale
DEBUG: Added attr 'calibscale' to channel 'intensity_green'
DEBUG: Found match: intensity_blue and intensity_calibscale
DEBUG: Added attr 'calibscale' to channel 'intensity_blue'
DEBUG: Found match: intensity_clear and intensity_calibscale
DEBUG: Added attr 'calibscale' to channel 'intensity_clear'
DEBUG: Found match: intensity_red and intensity_calibscale
DEBUG: Added attr 'calibscale' to channel 'intensity_red'
DEBUG: Found match: intensity_green and intensity_integration_time
DEBUG: Found match: intensity_blue and intensity_integration_time
DEBUG: Found match: intensity_clear and intensity_integration_time
DEBUG: Found match: intensity_red and intensity_integration_time
DEBUG: Found match: intensity_green and intensity_integration_time
DEBUG: Added attr 'integration_time' to channel 'intensity_green'
DEBUG: Found match: intensity_blue and intensity_integration_time
DEBUG: Added attr 'integration_time' to channel 'intensity_blue'
DEBUG: Found match: intensity_clear and intensity_integration_time
DEBUG: Added attr 'integration_time' to channel 'intensity_clear'
DEBUG: Found match: intensity_red and intensity_integration_time
DEBUG: Added attr 'integration_time' to channel 'intensity_red'
DEBUG: Added device 'iio:device0' to context 'local'
dev 'tcs3472', channel 'intensity_clear' (input, index: 0, format: le:U16/16>>0), found 3 channel-specific attributes
dev 'tcs3472', channel 'intensity_red' (input, index: 1, format: le:U16/16>>0), found 3 channel-specific attributes
dev 'tcs3472', channel 'intensity_green' (input, index: 2, format: le:U16/16>>0), found 3 channel-specific attributes
dev 'tcs3472', channel 'intensity_blue' (input, index: 3, format: le:U16/16>>0), found 3 channel-specific attributes
dev 'tcs3472', channel 'timestamp' (input, index: 4, format: le:S64/64>>0), found 0 channel-specific attributes
```

```
$ sudo iio_info 
Library version: 0.21 (git tag: a7cc0ce)
[... 74 lines cut ...]
				attr  1: integration_time value: 0.614400
				attr  2: raw value: 181
			intensity_green:  (input, index: 2, format: le:U16/16>>0)
			3 channel-specific attributes found:
				attr  0: calibscale value: 1
				attr  1: integration_time value: 0.614400
				attr  2: raw value: 191
			intensity_blue:  (input, index: 3, format: le:U16/16>>0)
			3 channel-specific attributes found:
				attr  0: calibscale value: 1
				attr  1: integration_time value: 0.614400
				attr  2: raw value: 160
			timestamp:  (input, index: 4, format: le:S64/64>>0)
		3 device-specific attributes found:
				attr  0: calibscale_available value: 1 4 16 60
				attr  1: current_timestamp_clock value: realtime

				attr  2: integration_time_available value: 0.002400 0.004800 0.007200 0.009600 0.012000 0.014400 0.016800 0.019200 0.021600 0.024000 0.026400 0.028800 0.031200 0.033600 0.036000 0.038400 0.040800 0.043200 0.045600 0.048000 0.050400 0.052800 0.055200 0.057600 0.060000 0.062400 0.064800 0.067200 0.069600 0.072000 0.074400 0.076800 0.079200 0.081600 0.084000 0.086400 0.088800 0.091200 0.093600 0.096000 0.098400 0.100800 0.103200 0.105600 0.108000 0.110400 0.112800 0.115200 0.117600 0.120000 0.122400 0.124800 0.127200 0.129600  [... cut ...] 566400 0.568800 0.571200 0.573600 0.576000 0.578400 0.580800 0.583200 0.585600 0.588000 0.590400 0.592800 0.595200 0.597600 0.600000 0.602400 0.604800 0.607200 0.609600 0.612000 0.614400
		2 buffer-specific attributes found:
				attr  0: data_available value: 0
				attr  1: watermark value: 1
ERROR: checking for trigger : Input/output error (5)
```

I've also tried modifying iio_attr to print whether the channel is enabled using the same function the bindings use and that returns without errors.

mhennerich, 2020-12-14:
In case there is an issue, it's likely in the iio pythom bindings.
Can you run iio_genxml on the target and provide it's output?
In theory we should be able to remotely create a python context from it and debug it.

XECDesign, 2020-12-14:
Sure thing. Thank you very much for taking a look. I spent some time banging my head against it and concluded that it must be a python bindings issue as well, but that's not an area I'm familiar with, so I wasn't getting very far.

```
$ iio_genxml 
DEBUG: Added attr 'in_intensity_calibscale' to device 'iio:device0'
DEBUG: Added attr 'raw' to channel 'intensity_green'
DEBUG: Added input channel 'intensity_green' to device 'iio:device0'
DEBUG: Added attr 'calibscale_available' to device 'iio:device0'
DEBUG: Added attr 'integration_time_available' to device 'iio:device0'
DEBUG: Added attr 'in_intensity_integration_time' to device 'iio:device0'
DEBUG: Added attr 'raw' to channel 'intensity_blue'
DEBUG: Added input channel 'intensity_blue' to device 'iio:device0'
DEBUG: Added attr 'current_timestamp_clock' to device 'iio:device0'
DEBUG: Added attr 'raw' to channel 'intensity_clear'
DEBUG: Added input channel 'intensity_clear' to device 'iio:device0'
DEBUG: Added attr 'raw' to channel 'intensity_red'
DEBUG: Added input channel 'intensity_red' to device 'iio:device0'
DEBUG: Added buffer attr 'watermark' to device 'iio:device0'
DEBUG: Added buffer attr 'data_available' to device 'iio:device0'
DEBUG: Add protected attr 'en' to channel 'intensity_blue'
DEBUG: Add protected attr 'en' to channel 'intensity_clear'
DEBUG: Add protected attr 'type' to channel 'intensity_blue'
DEBUG: Add protected attr 'en' to channel 'intensity_green'
DEBUG: Add protected attr 'index' to channel 'intensity_clear'
DEBUG: Add protected attr 'index' to channel 'timestamp'
DEBUG: Added input channel 'timestamp' to device 'iio:device0'
DEBUG: Add protected attr 'index' to channel 'intensity_red'
DEBUG: Add protected attr 'en' to channel 'intensity_red'
DEBUG: Add protected attr 'en' to channel 'timestamp'
DEBUG: Add protected attr 'index' to channel 'intensity_green'
DEBUG: Add protected attr 'type' to channel 'intensity_red'
DEBUG: Add protected attr 'type' to channel 'timestamp'
DEBUG: Add protected attr 'type' to channel 'intensity_green'
DEBUG: Add protected attr 'type' to channel 'intensity_clear'
DEBUG: Add protected attr 'index' to channel 'intensity_blue'
DEBUG: Found match: intensity_green and intensity_calibscale
DEBUG: Found match: intensity_blue and intensity_calibscale
DEBUG: Found match: intensity_clear and intensity_calibscale
DEBUG: Found match: intensity_red and intensity_calibscale
DEBUG: Found match: intensity_green and intensity_calibscale
DEBUG: Added attr 'calibscale' to channel 'intensity_green'
DEBUG: Found match: intensity_blue and intensity_calibscale
DEBUG: Added attr 'calibscale' to channel 'intensity_blue'
DEBUG: Found match: intensity_clear and intensity_calibscale
DEBUG: Added attr 'calibscale' to channel 'intensity_clear'
DEBUG: Found match: intensity_red and intensity_calibscale
DEBUG: Added attr 'calibscale' to channel 'intensity_red'
DEBUG: Found match: intensity_green and intensity_integration_time
DEBUG: Found match: intensity_blue and intensity_integration_time
DEBUG: Found match: intensity_clear and intensity_integration_time
DEBUG: Found match: intensity_red and intensity_integration_time
DEBUG: Found match: intensity_green and intensity_integration_time
DEBUG: Added attr 'integration_time' to channel 'intensity_green'
DEBUG: Found match: intensity_blue and intensity_integration_time
DEBUG: Added attr 'integration_time' to channel 'intensity_blue'
DEBUG: Found match: intensity_clear and intensity_integration_time
DEBUG: Added attr 'integration_time' to channel 'intensity_clear'
DEBUG: Found match: intensity_red and intensity_integration_time
DEBUG: Added attr 'integration_time' to channel 'intensity_red'
DEBUG: Added device 'iio:device0' to context 'local'
XML generated:

<?xml version="1.0" encoding="utf-8"?><!DOCTYPE context [<!ELEMENT context (device | context-attribute)*><!ELEMENT context-attribute EMPTY><!ELEMENT device (channel | attribute | debug-attribute | buffer-attribute)*><!ELEMENT channel (scan-element?, attribute*)><!ELEMENT attribute EMPTY><!ELEMENT scan-element EMPTY><!ELEMENT debug-attribute EMPTY><!ELEMENT buffer-attribute EMPTY><!ATTLIST context name CDATA #REQUIRED description CDATA #IMPLIED><!ATTLIST context-attribute name CDATA #REQUIRED value CDATA #REQUIRED><!ATTLIST device id CDATA #REQUIRED name CDATA #IMPLIED><!ATTLIST channel id CDATA #REQUIRED type (input|output) #REQUIRED name CDATA #IMPLIED><!ATTLIST scan-element index CDATA #REQUIRED format CDATA #REQUIRED scale CDATA #IMPLIED><!ATTLIST attribute name CDATA #REQUIRED filename CDATA #IMPLIED><!ATTLIST debug-attribute name CDATA #REQUIRED><!ATTLIST buffer-attribute name CDATA #REQUIRED>]><context name="local" description="Linux serge-pi400 5.4.80-v7l+ #5 SMP Tue Dec 1 20:01:54 GMT 2020 armv7l" ><context-attribute name="local,kernel" value="5.4.80-v7l+" /><context-attribute name="uri" value="local:" /><device id="iio:device0" name="tcs3472" ><channel id="intensity_clear" type="input" ><scan-element index="0" format="le:U16/16&gt;&gt;0" /><attribute name="calibscale" filename="in_intensity_calibscale" /><attribute name="integration_time" filename="in_intensity_integration_time" /><attribute name="raw" filename="in_intensity_clear_raw" /></channel><channel id="intensity_red" type="input" ><scan-element index="1" format="le:U16/16&gt;&gt;0" /><attribute name="calibscale" filename="in_intensity_calibscale" /><attribute name="integration_time" filename="in_intensity_integration_time" /><attribute name="raw" filename="in [... cut ...] ibute name="raw" filename="in_intensity_blue_raw" /></channel><channel id="timestamp" type="input" ><scan-element index="4" format="le:S64/64&gt;&gt;0" /></channel><attribute name="calibscale_available" /><attribute name="current_timestamp_clock" /><attribute name="integration_time_available" /><buffer-attribute name="data_available" /><buffer-attribute name="watermark" /></device></context>

DEBUG: Added device 'iio:device0' to context 'xml'
Context re-creation from generated XML succeeded!
```

mhennerich, 2020-12-14:
I pasted your context into a file called ctx1 -
That's what I get. Can you try to do the same on your end?

`Python 3.8.5 (default, Jul 28 2020, 12:59:40) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import iio
>>> ctx=iio.Context("xml:ctx1")
>>> device=ctx.find_device('tcs3472')
>>> device.find_channel('intensity_red').enabled
False
>>> device.find_channel('intensity_green').enabled
False
>>> device.find_channel('intensity_clear').enabled
False
>>> device.find_channel('intensity_blue').enabled
False
>>> device.find_channel('intensity_blue').enabled = True
>>> device.find_channel('intensity_blue').enabled
True
>>> device.find_channel('intensity_clear').enabled = True
>>> device.find_channel('intensity_clear').enabled
True
>>> device.find_channel('intensity_clear').id
'intensity_clear'`

XECDesign, 2020-12-14:
Same output as yours.
```
$ ./test.py 
DEBUG: Added device 'iio:device0' to context 'xml'
False
False
False
False
True
True
intensity_clear
```

mhennerich, 2020-12-14:
In case you use the default context?

XECDesign, 2020-12-14:
That looks promising. Works with the following:
```
ctx=iio.Context()
device = ctx.find_device('tcs3472')
```
but not with `device = iio.Context().find_device('tcs3472')`

tfcollins, 2020-12-14:
Can you run this with gdb and provide the stack trace when it segfaults?

-Travis

XECDesign, 2020-12-14:
```
Program received signal SIGSEGV, Segmentation fault.
0xb61db3f4 in iio_channel_is_enabled (chn=<optimized out>) at /home/pi/libiio/channel.c:449
449		return chn->index >= 0 && chn->dev->mask &&
(gdb) bt
#0  0xb61db3f4 in iio_channel_is_enabled (chn=<optimized out>) at /home/pi/libiio/channel.c:449
#1  0xb666dcc0 in ffi_call_VFP () from /usr/lib/arm-linux-gnueabihf/libffi.so.6
#2  0xb666e6cc in ffi_call () from /usr/lib/arm-linux-gnueabihf/libffi.so.6
#3  0xb668f0c8 in _ctypes_callproc () from /usr/lib/python3.7/lib-dynload/_ctypes.cpython-37m-arm-linux-gnueabihf.so
#4  0xb668f4ec in ?? () from /usr/lib/python3.7/lib-dynload/_ctypes.cpython-37m-arm-linux-gnueabihf.so
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) generate-core-file 
warning: target file /proc/9204/cmdline contained unexpected null characters
Saved corefile core.9204
```

[core.zip](https://github.com/analogdevicesinc/libiio/files/5690754/core.zip)

rgetz, 2020-12-15:
Thanks for the report/backtrace.

`iio_channel_is_enabled()` is crashing because channel is null.

```
448 bool iio_channel_is_enabled(const struct iio_channel *chn)
    449 {
    450         return chn->index >= 0 && chn->dev->mask &&
    451                 TEST_BIT(chn->dev->mask, chn->number);
    452 }
```

The question - is why `device = iio.Context().find_device('tcs3472')` fails, but `ctx=iio.Context(); device = ctx.find_device('tcs3472')` works.

boukeas, 2020-12-15:
> The question - is why device = iio.Context().find_device('tcs3472') fails, but ctx=iio.Context(); device = ctx.find_device('tcs3472') works.

This is why:
```
>>> import iio
>>> ctx = iio.Context()
>>> ctx.find_device('tcs3472').ctx
<weakref at 0xb6552180; to 'Context' at 0xb6715d70>
>>> iio.Context().find_device('tcs3472').ctx
<weakref at 0xb6552ae0; dead>
```
It looks like the `Context` created in the latter case is promptly destroyed, which makes sense if it is only referenced through weak references. 

But this doesn't explain why this works:
```
>>> iio.Context().find_device('tcs3472').find_channel('intensity_red').enabled
False
```
whereas this leads to a segmentation fault:
```
>>> iio.Context().find_device('tcs3472').find_channel('intensity_clear').enabled
```
Something is particularly wrong with the "clear" channel.

cristi-iacob, 2020-12-15:
Hi @boukeas and @XECDesign !

Can you please test if the modification from [this branch](https://github.com/analogdevicesinc/libiio/tree/py_ctx_ref_fix) solves the issue for you?

Thank you!

rgetz, 2020-12-16:
That resolves things for me (based on the tests I was running with @mhennerich context).

And it makes sense - it shouldn't be a weak reference...

XECDesign, 2020-12-18:
Sorry about the late response. The latest master branch does resolve the issue for me. Thank you. I'll take a look at updating the Raspberry Pi OS package as well.

closed 2020-12-18 by boukeas