analogdevicesinc/libiio #640: Support for rescanning/reopening devices in iiod

yaperez-anssi, 2020-12-22:
I'm currently playing with a PlutoSDR where I want to later expose some MMIO registers (from the PL/FPGA part). Since it seems that the PlutoSDR heavily relies on the IIO subsystem I'm trying to craft an IIO kernel module for that, so it can get exposed by libiio/iiod accross USB or network.

Right now I'm testing the driver using a kernel module so I don't have to rebuild and reflash the image (even to RAM) everytime.

I can load the kernel module and create a new IIO_SW_DEVICE device node which locally appears on the PlutoSDR, but is not exposed by iiod. I guess that's because iiod would need to rescan the devices, or be restarted. Unfortunately it seems that on the PlutoSDR it's not possible to restart iiod (it'll disappear from the network), so having a way to ask iiod for a rescan would be really nice.

rgetz, 2021-01-03:
This really isn't easy with the current architecture. I did have a branch that did something similar (looking for changes in sysfs), but it's complicated - since you can have multiple clients, and the backing iiod needs to be the same as the the client that is attached to it.

Your concept works in the simple use case, but not in the complicated ones (multiple connected clients)...

Let me think about it some more.

yaperez-anssi, 2021-01-04:
Good point, the clients might need to be notified so they can rescan themselves or something. I have to admit my issue might be specific to the PlutoSDR, because usually you might just want to restart the daemon (and maybe the clients), but unfortunately I can't do that.

rgetz, 2021-02-20:
There is no "notification" in the existing libiio implementation. All the client can do is close the context, and re-open things if it different. That's the way that the current implementation is architected. I think the tests I was doing would function, but would require a lot more testing... Let me think about it a bit more...

rgetz, 2022-10-20:
This is sort of a dup of #911 now.

yaperez-anssi, 2022-10-21:
Yes indeed, thanks for the pointer, I'll subscribe to that one.

rgetz, 2023-02-03:
This can be closed now (I can't anymore).

yaperez-anssi, 2023-02-03:
Closing because duplicate of #911

closed 2023-02-03 by yaperez-anssi