analogdevicesinc/libiio #644: iio_reg only addresses reg 0x0 when run on Pluto

tfcollins, 2021-01-05:
See post: https://ez.analog.com/adieducation/university-program/f/q-a/539472/plutosdr-v0-32-iio_reg-can-t-read-registers

I can repeat this on v0.32 firmware.

-Travis

rgetz, 2021-01-05:
I think Michael fixed that - what version of libiio is on that?

rgetz, 2021-01-06:
I can replicate on master, but not on the release.

mhennerich, 2021-01-11:
This was the fix -
https://github.com/analogdevicesinc/libiio/commit/f75ced2ab5690a3da72c2f83947a854a9f96871e

Can you try to specify the address in decimal instead of hex?

mhennerich, 2021-01-11:
Libiio version 0.21 was released Jun17, my fix was committed July31.
So our latest release is still broken.

# iio_reg ad9361-phy 3
0x5c
# iio_reg ad9361-phy 0x3
0x0
# 
# iio_info
Library version: 0.21 (git tag: v0.21)
Compiled with backends: local xml ip usb serial
IIO context created with local backend.
Backend version: 0.21 (git tag: v0.21)
Backend description string: Linux pluto 4.19.0-120601-g3a488163e40c #321 SMP PREEMPT Fri Dec 18 08:2l
IIO context has 8 attributes:
        hw_model: Analog Devices PlutoSDR Rev.A (Z7010-AD9363A)
        hw_model_variant: 0
        hw_serial: 1000002355237322002000160902163a3c
        fw_version: v0.32-dirty
        ad9361-phy,xo_correction: 40000000
        ad9361-phy,model: ad9363a
        local,kernel: 4.19.0-120601-g3a488163e40c
        uri: local:

rgetz, 2022-02-17:
I think this is fixed now.

```
# iio_reg ad9361-phy 3
0x5c
# iio_reg ad9361-phy 0x03
0x5c
# iio_reg ad9361-phy 0x3
0x5c
# iio_info 
Library version: 0.23 (git tag: v0.23)
Compiled with backends: local xml ip usb serial
IIO context created with local backend.
Backend version: 0.23 (git tag: v0.23)
```

closed 2022-02-17 by rgetz