analogdevicesinc/libiio #646: Do a review of the libserial timeouts

dNechita, 2021-01-21:
Verify which libserialport calls might timeout and handle the return value in case timeout occurs.
Some works has been done in #645 but only for call: sp_blocking_read_next().

rgetz, 2021-02-21:
There are only a few times that timeout is capable of being used in libserialport (that we use things)

```
    267 static ssize_t serial_write_data(struct iio_context_pdata *pdata,
    268                 void *io_data, const char *data, size_t len)
    269 {
    270         ssize_t ret = (ssize_t) libserialport_to_errno(sp_blocking_write(
    271                                 pdata->port, data, len, pdata->timeout_ms));
```

```
    283 static ssize_t serial_read_data(struct iio_context_pdata *pdata,
    284                 void *io_data, char *buf, size_t len)
    285 {
    286         ssize_t ret = (ssize_t) libserialport_to_errno(sp_blocking_read_next(
    287                                 pdata->port, buf, len, pdata->timeout_ms));
```

```
    309                 ret = libserialport_to_errno(sp_blocking_read_next(
    310                                         pdata->port, &buf[i], 1,
    311                                         pdata->timeout_ms));
```

and I think you have tried to capture things in those functions.

checking out: libserialport.h, its only these 4 functions which have timeouts:
https://www.sigrok.org/api/libserialport/0.1.1/a00004.html
```
enum sp_return 	sp_blocking_read (struct sp_port *port, void *buf, size_t count, unsigned int timeout_ms)
 	Read bytes from the specified serial port, blocking until complete. 
enum sp_return 	sp_blocking_read_next (struct sp_port *port, void *buf, size_t count, unsigned int timeout_ms)
 	Read bytes from the specified serial port, returning as soon as any data is available. 
enum sp_return 	sp_blocking_write (struct sp_port *port, const void *buf, size_t count, unsigned int timeout_ms)
 	Write bytes to the specified serial port, blocking until complete.
enum sp_return 	sp_wait (struct sp_event_set *event_set, unsigned int timeout_ms)
 	Wait for any of a set of events to occur. More...
```

it's easy to check to see where those 4 functions are used:
```
270:	ssize_t ret = (ssize_t) libserialport_to_errno(sp_blocking_write(
286:	ssize_t ret = (ssize_t) libserialport_to_errno(sp_blocking_read_next(
309:		ret = libserialport_to_errno(sp_blocking_read_next(
```

those are the 3 places where you did things in PR #645 and #647 

So, I think this can be closed?

dNechita, 2021-02-22:
Yes, I believe I covered all the functions that can timeout so I'm closing this.

closed 2021-02-22 by dNechita