analogdevicesinc/libiio #648: bindings: python: TypeError: 'Context' object is not callable

ekigwana, 2021-02-04:
Just tried running the pluto example from [pyadi-iio](https://github.com/analogdevicesinc/pyadi-iio) and got the following error: ` TypeError: 'Context' object is not callable`

https://github.com/analogdevicesinc/libiio/blob/359305cd482cbcd6a55386adc9a54b68364e7dad/bindings/python/iio.py#L1016

I replaced it with `self._ctx = device.context()` which appears to be what it is intended for and  moved the error to:

https://github.com/analogdevicesinc/libiio/blob/359305cd482cbcd6a55386adc9a54b68364e7dad/bindings/python/iio.py#L1331

Looking at the Context the is not definition to return a reference.

Not sure if the fix to replace all `.ctx()` with `.ctx` or in just some places or to define something in the Context

tfcollins, 2021-02-04:
Can you back out this change https://github.com/analogdevicesinc/libiio/commit/fe86c6b7c315674e3f7cad0cbac152504757bebe

-Travis

mrhgit, 2021-02-07:
This fix seems incomplete for two reasons:
 1) there are still calls to self.ctx() in the Device class, instead of self.ctx
 2) weakref does not protect the object from deletion, which is exactly what the Buffer class needs to do, because it uses the context object in the deletion routine (see the C code for buffer_destroy).

pcercuei, 2021-08-17:
@ekigwana @mrhgit Could you test my PR #735? Thanks!

pcercuei, 2021-08-18:
This should be fixed now thanks to #735. Closing the issue.

closed 2021-08-18 by pcercuei

ekigwana, 2021-08-18:
Late, but works for me now