analogdevicesinc/libiio #650: RX Timestamp Metadata

anilgurses, 2021-02-11:
Hi,
I am trying to retrieve RX timestamp metadata on ad9361 with libiio. However, it seems that lots of people have tried and failed on this. Are there any recent updates on this?

In the /examples/dummy-iiostream.c example there is a part that prints timestamp. 

```
// Print timestamp delta in ms
for (p_dat = iio_buffer_first(rxbuf, channels[channel_count-1]); p_dat < p_end; p_dat += p_inc) {
        now_ts = (((int64_t *)p_dat)[0]);
	printf("[%04" PRId64 "] ", last_ts > 0 ? (now_ts - last_ts)/1000/1000 : 0);
	last_ts = now_ts;
}
```

However, this is controlled by `has_ts` variable. That means, it checks for its support timestamp channel. I couldn't get anything from my ad9361. Isn't it possible to gather RX timestamp metadata with the current libiio?

Also related forum question I found on the analog devices website;

https://ez.analog.com/fpga/f/q-a/117408/adrv9361-rx-tx-timestamp-for-lte
https://ez.analog.com/linux-software-drivers/f/q-a/87641/fmcomms5-rx-timestamp
https://ez.analog.com/fpga/f/q-a/51587/timestamp
https://ez.analog.com/fpga/f/q-a/118200/adrv9361-adding-tx-timestamp

Do I need to modify HDL design and firmware?

ekigwana, 2021-02-12:
Yes, you need to modify the HDL design to your requirements.

rgetz, 2021-02-20:
specific questions about specific devices are better handled on ADI's EngineerZone. libiio is a generic library for many (any) IIO device.

closed 2021-02-20 by rgetz